portapack-mayhem/mayhem-firmware #9: Portapack and bad reception with screen turned on

DiscoMan18, 2020-04-30:
Hi,

I use combination of HackRF One and Portapack H2. Flashed your firmware and noticed that reception of local FM stations improves a lot when I turn off the screen of my portapack. With the screen on audio is barely audible and sinks in noise. On the other bands situation is better, but not much. 

Could you please check if you have these issues too and probably fix them somehow?

BR,

Alex.

helmarw, 2020-05-02:
do you have a USB cable connected to PC or batterypack.
both doesnt work very well, probably getting noise from the switching powersupply.
for me it only work when using the internal battery ...

DiscoMan18, 2020-05-02:
No, I'm getting these issues on battery power only. And turning the screen off helps to improve recepetion a lot.

eried, 2020-05-03:
Can you post a video with the effect? I do not notice a big change with the screen, but I notice some change in the signal when I plug the usb.

Dirk275, 2020-05-25:
Reception of HackRF with it's own telescope ant is highly variabel and depending on many things.
I watched it dependant on antenna counterbalance: if you have earphones connected, it makes a difference if you let their cable hanging down from HackRF (better reception) or upwards (worse reception).
So there are many things influencing ...
Reception usually gets better, if you ground HackRF.

eried, 2020-05-25:
@Dirk275 but do you experience bad reception with the screen on like @helmarw ?

Dirk275, 2020-05-25:
No, not really with H1.

eried, 2020-08-10:
I don't think there is much to add here. Feel free to reopen the issue if that´s not the case.

closed 2020-08-10 by eried