portapack-mayhem/mayhem-firmware #13: new update by jamesshao

ytsqkl, 2020-05-06:
https://github.com/jamesshao8/portapack-havoc-modified
1.he updated pocsag ,now you can choose p  or  n phase
2. add speaker control ,but h2 version will hotter hotter than h1
can you  merge into one project to keep everyone using the latest and most complete firmware.

nemanjan00, 2020-05-08:
Speaker control can be found in this PR: #17

eried, 2020-05-08:
Yes ofc I will merge those :) thanks to @nemanjan00  is only half to merge now.

@ytsqkl what do you mean with hotter and hotter? H1 is totally compatible

ytsqkl, 2020-05-09:
H2 has an external audio amplifier. Without using the speaker control icon, which is exactly the same as the H1。But the audio chip can be very hot by clicking on the Speaker Mute and then playing the speaker. The H1 is perfectly fine.

eried, 2020-05-09:
Oh I see, so I think the speaker wont be added to this fork because it might affect some users negatively. But I am going to integrate the other update

eried, 2020-05-09:
@ytsqkl I added your request 1) but 2) wont be added for now because as I said it might be negative for h2 users... this issue can be closed :)

ytsqkl, 2020-05-10:
Thank you very much indeed,that's i wanted, many hams in our country do not need the loudspeaker control, this will cause a lot of trouble, hope you can continue to develop more functions for Global Ham, thanks again.

closed 2020-05-28 by eried