portapack-mayhem/mayhem-firmware #21: Rotary encoder problem

nemanjan00, 2020-05-09:
Are you having problem where, while you scroll, it happens that encoders sometimes go wrong direction? 

https://imgur.com/a/vY2sXjT

nemanjan00, 2020-05-09:
This is the best I was able to come up, and is not perfect solution, fast switching of direction can be a problem: 

https://github.com/nemanjan00/portapack-havoc/commit/4f4999ddfd611870e54c90a92973fbb400fbf9e2

ytsqkl, 2020-05-10:
I also often encounter this problem ， hope  it to be solved

nemanjan00, 2020-05-10:
So far, I was able to reduce it, not to solve it completly... 

Will send PR, when I figure out what is the most effective solution

eried, 2020-05-10:
> I also often encounter this problem ， hope it to be solved

Which board you have? H2? Can you share a video of the issue?

nemanjan00, 2020-05-10:
@ytsqkl, I do have a hack to make this work better, but, it would cause some problems to people without this issue, so, it is not going to be merged. 

Only maybe as something that is optional to turn on in settings... 

Can you test my image and confirm it is working better? 

[image.zip](https://github.com/eried/portapack-havoc/files/4606191/image.zip)

ytsqkl, 2020-05-11:
> > I also often encounter this problem ， hope it to be solved
> 
> Which board you have @ytsqkl ? H2? Can you share a video of the issue?

H2

ytsqkl, 2020-05-11:
solved on my portapack

[Rotary.zip](https://github.com/eried/portapack-havoc/files/4609321/Rotary.zip)

eried, 2020-05-11:
It was @nemanjan00

ytsqkl, 2020-05-11:
> It was @nemanjan00

oh，sorry    @nemanjan00

eried, 2020-05-11:
How it was before @ytsqkl ? exactly the same as @nemanjan00  video? I have also the H2 but I do not experience that magic jumpyness

ytsqkl, 2020-05-11:
> How it was before @ytsqkl ? exactly the same as @nemanjan00 video? I have also the H2 but I do not experience that magic jumpyness

yes, the same as @nemanjan00 video

eried, 2020-05-11:
@nemanjan00 I think it would be good to add that option for the encoder fix. You can use the "Play dead" option in "UI" since it wont be active anymore in future versions

ytsqkl, 2020-05-12:
thanks  @eried and @nemanjan00

eried, 2020-05-12:
@nemanjan00 did a poll and very little people has this problem, they say it might be some dirt in the encoder, @ytsqkl can you try disconnecting the battery and putting a drop of alcohol or electronic cleaner solution to the rotary encoder to see if the problem disappears?

ytsqkl, 2020-05-12:
> @nemanjan00 did a poll and very little people has this problem, they say it might be some dirt in the encoder, @ytsqkl can you try disconnecting the battery and putting a drop of alcohol or electronic cleaner solution to the rotary encoder to see if the problem disappears?

My portapack was bought a month ago. It's almost new. It's been used at home. I haven't used it outdoors. I just cleaned rotary encoder with 95% alcohol. I flashshed the havoc firmware ，but the problem still exists. The firmware that nemanjan00 gave me doesn't have this problem. Maybe it's the quality problem of portapack produced by different manufacturers.

eried, 2020-05-12:
OK fine if you already tried to cleaned, the issue is maintaining a feature that only affects very little users:
[image]

@nemanjan00 has the idea of researching why it happens, maybe there is a missing cap in your boards

Anyhow, good that it is fixed for you

ytsqkl, 2020-05-12:
> OK fine if you already tried to cleaned, the issue is maintaining a feature that only affects very little users:
> [image]
> 
> @nemanjan00 has the idea of researching why it happens, maybe there is a missing cap in your boards
> 
> Anyhow, good that it is fixed for you

It is likely that this problem will cause signal burr

nemanjan00, 2020-05-12:
@ytsqkl feel free to use my fork of this project for now. 

I am going to keep it sync with this project, but, with a few changes to adapt it to my usage (just some changed default values, for example higher volume)

ytsqkl, 2020-05-13:
but 75 volume is too high for me 。。。。。

ytsqkl, 2020-05-13:
> @ytsqkl feel free to use my fork of this project for now.
> 
> I am going to keep it sync with this project, but, with a few changes to adapt it to my usage (just some changed default values, for example higher volume)

I will always focus on this project and your project, especially like portapack

closed 2020-05-28 by eried