portapack-mayhem/mayhem-firmware #29: can‘t remember mode and step

ytsqkl, 2020-05-11:
[image]
it can remember frequency and volume,but analog receive mode and frequence step cant't be saved and loaded on start

eried, 2020-05-11:
Yes, I agree it would be nice. But it might need to work everywhere, sort of: Save environment, and Restore to default, etc.

ytsqkl, 2020-05-12:
> Yes, I agree it would be nice. But it might need to work everywhere, sort of: Save environment, and Restore to default, etc.
 
most hams need  frequency 、frequency step 、bw（like DSB,USB,LSB .......）、volume remembered， it would be nice。

eried, 2020-05-12:
I agree with any enhancement :D there is some persistant memory, and also helpers for using it. You can give it a go starting with simple stuff like remembering last volume in all apps and send a PR

ytsqkl, 2020-05-12:
Unfortunately, I don't have any experience in C programming. I can only use C# programming，but i'm learning

eried, 2020-05-12:
if is fairly similar, take a look. I do not think people really use C/C++ by desire xD, I also prefer newer languages. The radio part is quite confusing but the UI is very cleanly implemented. If you find the location of the default volume, try to check how the values from settings are being save and voilá :D

ytsqkl, 2020-05-13:
> if is fairly similar, take a look. I do not think people really use C/C++ by desire xD, I also prefer newer languages. The radio part is quite confusing but the UI is very cleanly implemented. If [...]

yeah