portapack-mayhem/mayhem-firmware #30: Thank you for your firmware, can you add WFM bandwidth adjustment function

vfatman0825, 2020-05-12:
Hello eried, this is BG4QL, I really like the new UI and fonts you made. Can I add a function to adjust the bandwidth in WFM mode?

eried, 2020-05-12:
Sure thing but first I want to polish the UI a bit more for a release. I have seen the bandwidth being set in the audio app, maybe you could experiment with it meanwhile

vfatman0825, 2020-05-12:
> Sure thing but first I want to polish the UI a bit more for a release. I have seen the bandwidth being set in the audio app, maybe you could experiment with it meanwhile

Thank you very much for your reply. I hope WFM can be adjusted to 38KHz so that I can try to receive NOAA signals. I have been communicating with Shao. He is also very interested in this. If you have a wechat, you can discuss it with us. Thank you again for your UI improvements, and again for your H2 case

eried, 2020-05-12:
Thanks! Shao is way more expert in tunning up the difficult-to-understand parts, I am just working over the annoying stuff so the end product feels more professional (in my personal opinion ofc). I do not have wechat.

gullradriel, 2023-05-10:
Bandwidth adjustments done.
Update to latest stable or nightly.

closed 2023-05-10 by gullradriel