portapack-mayhem/mayhem-firmware #35: H1 stuck on capture, stuck on OOK

bl4k1st, 2020-05-26:
I have a Portapack H1 and a found two problems.

1) When I capture something and leave the capture mode without stopping capture, the HackRF locks up. I am stuck with a capture screen and a 0B file. I have to hit the reset button.

2) When going to transmit-OOK, I can set the type with the rotary. When rotating to "12E" I can go back again. When rotating from "12E" one to the right the Portapack hangs up. I cannot go back I have to hit the reset button.

eried, 2020-05-26:
Thanks for the report, I will take a look. If they happen also on the H2 I will try to fix them.

sliceratwork, 2020-05-26:
@bl4k1st , have a look at [z4ziggy's fix](https://github.com/z4ziggy/portapack-havoc/commit/395bc486f12b4303cadfc2fa59e9095a409aaaa8).

eried, 2020-05-26:
Awesome! I will use that fix, it will be available on next release but for now is on the last nightly: 
https://github.com/eried/portapack-mayhem/releases/tag/nightly

and @bl4k1st 1) is halting for me too, so I will research the capture app

eried, 2020-05-28:
OK, both problems are fixed now on the latest nightly https://github.com/eried/portapack-mayhem/releases/tag/nightly

It will be included on 1.0.3 in the future.

eried, 2020-06-04:
I am closing this issue because it seems it is solved (yes?)

closed 2020-06-04 by eried

bl4k1st, 2020-06-05:
yes, should be fixed, thank you :)