portapack-mayhem/mayhem-firmware #37: ADS-B - ICAO list issue

Dirk275, 2020-05-28:
ADS-B world map now looks really fine. Thanks!

There is a small problem in the UI of the ICAO table:
If there are several entries (nearby planes) with some of them getting out of reach, then there are "empty" entries between still active entries.
If the cursor is positioned over an active entry at the end of the list, then it is not possible to move the cursor up jumping over inactive entries.
In this situation it is also impossible to get out of the app with <-, because the cursor can't reach the title line.

Additional improvement question: is it possible to decode the track/heading infos and turn the plane symbol on the map to that direction?

eried, 2020-05-28:
Hi, I will take a look on the bug. About the improvement, I guess it is possible but it needs someone to poke on it :D maybe it is a chance for you to [explore](https://github.com/eried/portapack-mayhem/wiki/How-to-colaborate#coding-new-stuff-or-fixing-bugs) the source.

Dirk275, 2020-05-28:
Thanks for looking on the possible bug.

I made first steps on looking at the code, but it is too  complex for me.
I'm really interested in adding/improving, but the entry level is relatively high. I am programming uCs in GCC and in the Arduino environment and already wrote some simple libraries for sensors, servos, robots, software decoders for Morse code, GPS data and DCF ...
But if I look at such a complex FW with functions spread over multiple cpp files, I tend to give up before beginning. 
So of course I don't expect that you do everything here, but think I'm actually not very useful ... but I will read the linked page first and try to learn.

eried, 2020-05-28:
Try to find who draws the icon you want to rotate. First make it point to a different bearing. I don't know either any details but I start poking around like that.

eried, 2020-05-28:
I do not have much airplane activity to test nearby. Can you record a video explaining the problem? Please use the latest bin from https://github.com/eried/portapack-mayhem/releases/tag/nightly

Dirk275, 2020-05-29:
About the improvement:
Only looking at the code I think the function GeoMap::draw_bearing is responsible for painting the red triangle in the screen Center pointing East (?).
But this function is always called with constant parameters, especially for angle. The function itself already seems to be able to draw the triangle pointing to different directions depending on the plane's heading, but needs parameter angle.

About the bug:
I will try to make a video, but in the moment the times of heavy air traffic activity are rare here also. The described app behaviour is not new: it was the same before in last havoc FW.

eried, 2020-05-29:
Try to change the default one to see if it works first. Go with little changes, there is a lot of undocumented functionality (maybe the angle does not work)

With the bug: Clearly is the same, nothing has changed in the code there, but since I do not use the app, video helps to understand it and start poking.

Dirk275, 2020-05-31:
About improvement:
The mentioned draw_bearing function is called in GeoMap::paint also with an angle_ parameter, but I can see no decoding of heading/track anywhere. The only infos decoded from ADS-B are ICAO, callsign, altitude, lat, lon.
So I think the track is not decoded. One possibility to show the plane's movement on the map would be to draw a permanent line along the plane's track, that overlays the map. Would be nice ...

About bug:
Video not possible due to low traffic and fast changing display contents.
What I can describe: If I come back to the ICAO list from map view, then the cursor is stuck on the entry of the last plane watched on the map.
Possibly only positioning the cursor on the baseband after returning from map view would fix the problem or a part of the problem.

eried, 2020-05-31:
You can save a second position and calcule the bearing no?

Dirk275, 2020-05-31:
Yes, should be possible.

ArjanOnwezen, 2021-11-11:
Looks like the issue is solved with pr #422

closed 2021-11-12 by eried