portapack-mayhem/mayhem-firmware #39: Freezing when running wave viewer

gregoryfenton, 2020-05-30:
**Bug**
The system is freezing when I click on a wave file in Wave Viewer

**To Reproduce**
Steps to reproduce the behavior:
1. Click on Tools - Wave Viewer - Open - Browse to a wave file on SD card
2. Click on a wave file

**Expected behavior**
A view of the wave file

**Actual behaviour**
The system becomes unresponsive, taking a reset to fix.

**Additional**
Video showing issue attached

[20200529-221240_H61TOKdu_n2GA.zip](https://github.com/eried/portapack-mayhem/files/4705475/20200529-221240_H61TOKdu_n2GA.zip)

eried, 2020-05-30:
I confirm the hang, can you test different wav formats? Do you know if this ever worked?

gregoryfenton, 2020-05-30:
I remember it working a while back, but unsure of the firmware or whether it was on havoc or PortaPack base.
I'll generate a bunch of different formats to test and see if that makes a difference.

eried, 2020-05-30:
Thanks! that help a lot, if you can share those files so I can test old firmwares until I find one that works to discover what is the problem

gregoryfenton, 2020-05-30:
The following files are from https://en.wikipedia.org/wiki/WAV
11k16bitpcm.wav works
11k8bitpcm.wav freezes
11kadpcm.wav freezes
11kgsm.wav freezes
11kulaw.wav freezes
8k16bitpcm.wav works
8k8bitpcm.wav freezes
8kadpcm.wav freezes
8kcelp.wav works
8kgsm.wav freezes
8kmp316.wav freezes
8kmp38.wav freezes
8ksbc12.wav works
8ktruespeech.wav freezes
8kulaw.wav freezes

So  it appears that

eried, 2020-05-30:
I think the only format it should support is the format captured  by the audio app and the soundboard app

gregoryfenton, 2020-05-30:
Agreed, but the files in /WAVE/ should be playable in wav viewer.

eried, 2020-05-30:
yes, those are some examples I created for the soundboard, I will check take a look during the week why it does not work

alain00091, 2020-08-06:
I found the problem...

But first you need to know that today it supports only ONE CHANNEL/16 bits format, so I do not know why in the SDCARD basic files they are all unsupported format ...

Second Yes there is a bug because it detects that the format is not supported but does not print it on the screen and stay blocked
Same issue if you select a file, pull out the SDCARD and try to open it

Files are not playable right now, only viewable with time and scale.

I fixed it and works well, I'm opening a PR

eried, 2020-08-06:
Very cool! Thanks. But yes it is weird that the same recordings aren't supported

alain00091, 2020-08-06:
[WAV-OK.zip](https://github.com/eried/portapack-mayhem/files/5036547/WAV-OK.zip)
[WAV-NOT-OK.zip](https://github.com/eried/portapack-mayhem/files/5036550/WAV-NOT-OK.zip)

Hello Eried,  
Nice to see you again
Here I give you some examples of files , good ones and bad ones   the amazing thing is that the one who are NOT working are provided with the code ....

I can debug the code quicker now with my DEBUG feature, I'm still working on that but if you like it please promote it
I do not know how the other designers are debugging the code,( I would like to know how) , I'm using an old method (from printf linux/unix) but it works !
Cheers

eried, 2020-08-06:
Hi @alain00091 I think you are a bit confused on how PR work. Please check https://www.youtube.com/watch?v=cYLzmMFe7-E on how to make contributions. You are mixing all commits in one.

Check the video and create a small PR with the wav fix only :)

eried, 2020-08-10:
This was kinda solved by one of the latest PR. An error message should appear in the next versions instead of the full freeze.

closed 2020-08-10 by eried