portapack-mayhem/mayhem-firmware #43: SCANNER Improvement (AM DSB/USB/LSB, NFM and WFM)

alain00091, 2020-06-02:
Hello Eried,
I made  3 new scanners entries in navigation menu ( AM(dsb,usb,lsb), NFM and WFM) with 3 differents default files : SCANNER_AM, SCANNER_FM and SCANNER_NFM.
I have to test it again with your new code (1.0.3) in about 2/3 days 
I added also a step entry in the files (for each RANGE) for am,fm, and nfm depending about the country (US/EUROPE) + default step if there is no step
I spend a lot of time to fix/make improvement in freqman.cpp : no check about maximum frequency supported by hackrf_one, frequency a: but not b:, frequency b; but not a:,b: frequency b: lower than a;,...

If you or someone else can give me default frequencies for AM/NFM when the files SCANNER_AM/SCANNER_NFM doesn't exists I will appreciate, it is also depending about the country,

eried, 2020-06-02:
Great @alain00091 any contribution is well received :)

Oh I have no idea of the defaults. What kind of improvement did you made to freqman besides the maximum? 

I changed the buttons (removing exit) and organizing stuff a bit, but I found some other bugs that I will need to fix first

alain00091, 2020-06-03:
I checked freqman a lot because I add problems with scanner file: if you mix and match direct freq (f=) and range freq (a= + b= ) in the same file it is not well read !
[SCANNER_AM_2.TXT](https://github.com/eried/portapack-mayhem/files/4724851/SCANNER_AM_2.TXT)
[SCANNER_AM_1.TXT](https://github.com/eried/portapack-mayhem/files/4724852/SCANNER_AM_1.TXT)
The one starting with f= as the first line is not well understood, the other starting with a= and b= is ok , why ?  , I fixed it this morning
I added different check from any file in /FREQMAN ( not only my new scanner files):
1 Frequency f=/a=/b= within the limits   error if =0, negative or more than 7500000000
2 lines with special characters  0x0d ( end of line from DOS)
3 empty lines
4 lines without f:/a:/b:d:
5: if a=  > b= error
6 if a=exists but not b=  error and the reverse
7 if a= and b= exists but no description      error
8 if there is too many lines           error
9 description too long                 error

I added 
comment line like C    starting with "//" (will not be putted in the database)
and   if it is a Range, optionally  we can add a step (0,1,2,3,4,..)  for AM/NFM/FM
    for AM  ( 9Khz US/10Khz), FM (100k/50K in some countries)
enum freqman_entry_step {
	STEP_DEF = 0,						// default
	AM_US,								// 10 Khz   AM/CB
	AM_EUR,								// 9 Khz	LW/MW
	NFM_1,								// 12,5 Khz (Analogic PMR 446)
	NFM_2,								// 6,25 Khz  (Digital PMR 446)
	FM_1,								// 100 Khz
	FM_2,								// 50 Khz
	ERROR_STEP

and I can add more...
so for CB a=27000000,b=27050000,d=CB,s=1    means 10Khz steps until the b=
I will continue tomorrow and update you

alain00091, 2020-06-08:
[code.zip](https://github.com/eried/portapack-mayhem/files/4744527/code.zip)
The source code is based on your release 1.0.3 , I rename it to 1.0.4 ....
the new scanner files are SCANNER_AM,..;, I do not read anymore SCANNER.TXT
[scanner-new.zip](https://github.com/eried/portapack-mayhem/files/4744541/scanner-new.zip)
I give you my test files
[Test-files.zip](https://github.com/eried/portapack-mayhem/files/4744549/Test-files.zip)

I check ( with freqman.cpp)      
minimum and maximum frequency, in range if a= is bigger than b=
empty line
line without f= or a= or b=
line with EOL from windows ( 0x0D/0X0A)
Range without description
description too long
line too long
Step error
too many lines ( as before : limit is 99 lines per file)

I added comment lines   starting with "//"  

Error line + comment line are printed only if  at top of file it starts with //DEBUG like a comment

There was 2 types of entry in the database : ( SINGLE/RANGE), I added a third one (ERROR)

If you have ideas, if you find bugs , tell me
Cheers

eried, 2020-06-08:
Can you create a pull request for the code? here is a guide: https://github.com/eried/portapack-mayhem/wiki/How-to-collaborate#coding-new-stuff-or-fixing-bugs

It would be a never ending task if I have to check zip files with code.

alain00091, 2020-06-08:
Ok Eried, I can do it but if I make a new fork it will be a new version ?  so there will be havoc, mayhem, and mine , no ? is it not possible to work together ...

eried, 2020-06-08:
If you create a fork, you can do pull request so you help the main project and everyone gets the code you are working on :)

eried, 2020-06-08:
It will duplicate the project, but that is how it works, you need a copy of the project to work