portapack-mayhem/mayhem-firmware #44: Icons disappear when using the jog wheel to move the selection

gregoryfenton, 2020-06-02:
**Describe the bug**
White icons disappear when the jog wheel is used to navigate the menu system

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Debug menu
2. Move the scroll wheel to highlight different options such as Memory, SD Card
3. Notice the icon on the currently selected item disappears

**Expected behaviour**
The icon should still be present.

**Additional**
This only happens where the icon is purely white with no other colours. Other menus have coloured icons which do not show this behaviour.

**Suggested fix**
Change icons to something other than white, or invert the icon colours on rollover. This is already the case with the menu item text which changes from white to black when selected.

eried, 2020-06-03:
Hi, is this on the release or nightly?

gregoryfenton, 2020-06-03:
This is on the nightly builds, release 1.0.3 has a purely text based menu for the Debug menu.
It was introduced in [2cd2443](https://github.com/eried/portapack-mayhem/commit/2cd2443eb25b83db23a95b22206aca5ec8fcfc11)

The issue is using
ui::color::White()
which gives a white icon on a white background when that option is selected.
You may wish to use
ui::color::Green()
for Memory, Peripherals, Temperature as they are all components mounted on a PCB which is usually green,
and
ui::color::Black()
for SD Card and Controls as SD cards are more often than not black, and the jog wheel on my PortaPack is black.

I am currently cloning your repository locally to submit a pull request with the suggested changes.

eried, 2020-06-03:
Cool, thanks :D

gregoryfenton, 2020-06-03:
My patch works and builds.
[image]

eried, 2020-06-03:
Cool!! I need to make new icons tho. The menu was patched by @euquiq

euquiq, 2020-06-03:
Nice catching! thanks!

eried, 2020-06-03:
Im not sure 🧐 the were on the original portapack, then removed on havoc.

eried, 2020-06-03:
BTW the standard coloring of the submenues in havoc is green for complete ones, yellow for half, grey for non implemented ones. I'm going to draw some cool icons and then check how the coloring fits

euquiq, 2020-06-03:
> BTW the standard coloring of the submenues in havoc is green for complete ones, yellow for half, grey for non implemented ones. I'm going to draw some cool icons and then check how the coloring fits

A thought: I've seen there is also dark_cyan() used in options menu, maybe to distinguish / settings / tools and eventually debug menu items ? (as in "this is no app in the usual sense") ?

eried, 2020-06-03:
I used @gregoryfenton PR as a placeholder for some quick icons and colors meanwhile. I will try later how it looks with dark_cyan or his colors, but for artistic "decision" would be nice to have some poll system because that's mostly opinion based.

gregoryfenton, 2020-06-03:
Looking at the code, there are not many places with nullptr instead of an icon (omitting commented lines):
firmware/application/apps/ui_script.cpp:		menu_view.add_item({ freqman_item_string(frequencies[n]), ui::Color::white(), nullptr, [this](){ on_frequency_select(); } });
firmware/application/apps/ui_settings.cpp:		{ "Radio",			ui::Color::dark_cyan(), nullptr,	[&nav](){ nav.push<SetRadioView>(); } },
firmware/application/apps/ui_settings.cpp:		{ "UI", 			ui::Color::dark_cyan(), nullptr,	[&nav](){ nav.push<SetUIView>(); } },
firmware/application/apps/ui_settings.cpp:		{ "Date/Time",		ui::Color::dark_cyan(), nullptr,	[&nav](){ nav.push<SetDateTimeView>(); } },
firmware/application/apps/ui_settings.cpp:		{ "Touch screen",	ui::Color::dark_cyan(), nullptr,	[&nav](){ nav.push<TouchCalibrationView>(); } },
firmware/application/apps/ui_debug.cpp:		{ "RFFC5072",    ui::Color::white(),	nullptr,	[&nav](){ nav.push<RegistersView>(
firmware/application/apps/ui_debug.cpp:		{ "MAX2837",     ui::Color::white(),	nullptr,	[&nav](){ nav.push<RegistersView>(
firmware/application/apps/ui_debug.cpp:		{ "Si5351C",     ui::Color::white(),	nullptr,	[&nav](){ nav.push<RegistersView>(
firmware/application/apps/ui_debug.cpp:		{ audio::debug::codec_name(), ui::Color::white(),	nullptr,	[&nav](){ nav.push<RegistersView>(
firmware/application/ui_navigation.cpp:		{ "Wave viewer",	ui::Color::blue(),		nullptr,				[&nav](){ nav.push<ViewWavView>(); } },
firmware/application/ui_navigation.cpp:		{ "Antenna length",	ui::Color::yellow(),	nullptr,				[&nav](){ nav.push<WhipCalcView>(); } },
firmware/application/ui_navigation.cpp:		{ "Wipe SD card",	ui::Color::red(),		nullptr,				[&nav](){ nav.push<WipeSDView>(); } },

I am happy to work through them and find suitable icons for these lines, I  could probably have done it in the time it took to type this response :P

gregoryfenton, 2020-06-03:
As for colours, it's your fork. You have creative control :D

euquiq, 2020-06-03:
I am just finishing compiling and testing my new "version" (with the speaker control in settings and status bar). I put the menues in dark_cyan ... I am also converting the debug->peripherals into a button based menu ... 

It will be interesting to visualize the color ... but also, it is clear that we need more icons :)

eried, 2020-06-03:
> As for colours, it's your fork. You have creative control :D

I want what looks the best 😎, and looks are oftenly based on "popularity".

> I am just finishing compiling and testing my new "version" (with the speaker control in settings and status bar). I put the menues in dark_cyan ... I am also converting the debug->peripherals into a [...]

Nice! This little details make the whole firmware to feel more pro

eried, 2020-06-04:
I am closing this issue because it seems it is solved (yes?)

closed 2020-06-04 by eried

gregoryfenton, 2020-06-07:
I'd say yes, it is solved.
Also, love the new icons and dark_cyan looks great :)