portapack-mayhem/mayhem-firmware #45: H1 Screen turns off when transmitting

gigawatts, 2020-06-03:
**Describe the bug**
In all of the transmit modes I tried in FW v1.0.3 on my PortaPack H1, the screen backlight turns off when transmission starts.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any transmit mode
2. Tap on Start
3. Screen turns off
4. Screen turns back on when any physical key is pressed. No response to touch screen taps.

**Expected behavior**
Transmission start/stop does not turn screen off.

eried, 2020-06-03:
Have you checked the upper switches? Specially the one that looks like a Spy 🕵️‍♂️

gigawatts, 2020-06-03:
Ha, that did it. Thanks!

Can I make a suggestion? In the Settings > UI menu, add toggles with descriptions for each of those icons across the top, and their current state.

Or even just a quick reference legend in settings or a help section somewhere, so we can understand what those icons mean without searching a wiki somewhere. A list of features, and a short description of each, might be a very helpful "field guide". For instance, "BHT Xy/EP" with a little shower icon, I have absolutely no idea what that means or what it is used for by just looking at it.

eried, 2020-06-03:
Lol me neither, but anyone can colaborate on those things you said, documentation or code https://github.com/eried/portapack-mayhem/wiki/How-to-colaborate

gigawatts, 2020-06-03:
Does the firmware offer a bitmap/png file viewer? If it does, you could provide a very simple set of images one could just toss on the SD card as a field guide and view from the file manager.

eried, 2020-06-03:
I am not completely sure of that. There is capability of reading bmp (inside the source) though but probably not other more useful formats and it would required a very dedicated contributor to make an app and the decoders. Probably the simplest would be to open simple bmps from the file manager.

gigawatts, 2020-06-03:
And I realize I'm way off topic now, but it might not hurt to add a "this project is no longer active, please go over here" message to the furrtek/portapack-havoc readme.md. I was quite confused for probably 20 minutes last night before I dug into the commit history to see why the firmware/ blah.bin file was no where to be found, as all the old instructions still pointed to that now broken link. I saw your comment on the question asked of the commit that removed the file, then found this repo. I would expect most to get lost far before that point.

gigawatts, 2020-06-03:
Anyway, thanks for the discussion. I'll have to dig into the code a bit deeper, see if there is anywhere I can contribute. Thanks for helping to keeping this hardware alive and more and more useful!

closed 2020-06-03 by gigawatts