portapack-mayhem/mayhem-firmware #46: Write the developer manual

eried, 2020-06-03:
Including details about the main parts of the firmware:
- [x] Persistent Memory
- [ ] UI flow (switching screens)
- [x] Compilation
- [ ] Tools (scripts and small tools to create bitmaps, etc)
- [ ] Widgets
- [ ] Top toolbar

Finally, some guides are necessary for helping newcomers:
- [ ] Create a simple graphical app
- [ ] Create a complex app (with some custom demodulation)

eried, 2020-06-03:
@euquiq what do you think about renaming the Persistent Memory wiki entry and writing a small intro how it work? I think it is an important piece.

euquiq, 2020-06-03:
Sure, I can try ! Right now I am finishing implementing  a working "speaker" functionality for H1 models, and it is also helping me understanding the code, which in turn will help on adding to the wiki, particularly on the Persistent Memory. Stay tuned ...

euquiq, 2020-06-04:
I added a bit more of information on the now renamed wiki "Persistent Memory" section

eried, 2020-06-04:
> I added a bit more of information on the now renamed wiki "Persistent Memory" section

Looks very nice!

notpike, 2022-01-04:
Hi,

Still figuring out the codebase however I'm adding in details for [Create-a-simple-app](https://github.com/eried/portapack-mayhem/wiki/Create-a-simple-app). I foresee this Wiki page becoming a little bolted from what I'm adding but it may be worth while to split this up a little.

eried, 2022-01-05:
Yes maybe you could move some stuff to "create a more complex app" and the simple one is a brief introduction. And we close this issue :)

notpike, 2022-01-06:
Hi,

I just finished up with widgets and added an example "Hello World" in [Create-a-simple-app](https://github.com/eried/portapack-mayhem/wiki/Create-a-simple-app). This is a rough draft so if you like to see any changes please let me know.

eried, 2022-01-07:
Looks very nice, I think we can close this issue now, just for the sake of closing issues and to release the small bounty :)

closed 2022-01-07 by eried