portapack-mayhem/mayhem-firmware #51: Docker - compilation problems ?

euquiq, 2020-06-05:
Eried, I am new to docker too and I cannot understand if this error (or lack of progress in the compilation) is mine, or there is something funny going on with the docker image.

Up until a couple of hours ago, issuing the command:

`docker run -it -v ${PWD}:/havoc portapackccache`

Whould compile everything and give me the ready to upload `portapack-h1_h2-mayhem.bin`

But now, it starts compiling, and just stop (at almost the beginning of the process), as shown below, with no error:

```
... 
[ 37%] Built target baseband_rds.elf
[ 37%] Built target baseband_adsbrx.elf
[ 37%] Built target baseband_nrfrx.elf
[ 38%] Built target baseband_audio_tx.elf
[ 38%] Built target baseband_mic_tx.elf
[ 40%] Built target baseband_adsbtx.elf
[ 40%] Built target baseband_capture.elf
[ 40%] Built target baseband_pocsag.elf
[ 41%] Built target baseband_jammer.elf
[ 41%] Built target baseband_nfm_audio.elf
[ 41%] Built target baseband_afsktx.elf
[ 47%] Built target baseband
[ 56%] Built target hackrf_usb_dfu_objects
[ 57%] Built target hackrf_usb_dfu.elf
[ 57%] Built target hackrf_usb_dfu.bin
dfu-suffix (dfu-util) 0.8

Copyright 2011-2012 Stefan Schmidt, 2013-2014 Tormod Volden
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Suffix successfully added to file
[ 57%] Built target hackrf_usb.dfu
[100%] Built target application.elf
[100%] Built target application
[100%] Built target firmware
enrique@MBP-de-Enrique portapack-mayhem % 
```

eried, 2020-06-05:
Are you sure it is not outputting it? There is no errors

euquiq, 2020-06-05:
Weird ... All I can say is that after the `[100%] Built target firmware` It finished and returned into my command prompt. When peekign into Docker dashboard, the machine was off.

Having said that, **NOW I did reboot my computer, and it is working OK, as expected.** 

Hence I am closing this issue. Maybe I was too fast to declare an issue :)

closed 2020-06-05 by euquiq