portapack-mayhem/mayhem-firmware #53: Backing up current firmware

scriptitall, 2020-06-06:
Hello, I have looked everywhere I can think of, but have been unable to find an answer as to how I can download the currently installed fireware from the portapack. I want to make a backup before flashing the mayhem firmware, in case I want to revert for whatever reason.

eried, 2020-06-06:
You will not want to revert to a previous version :)

I am not sure if there is a way to backup it, but I am 99% sure that you have one of these [versions](https://github.com/eried/portapack-mayhem#what-about-havocgridrfjamesshao8jboones) that are easy to recover if you want some old-school firmware.

scriptitall, 2020-06-07:
I'm trying to update firmware to mayhem, but the updater batch script isnt working. It says no HackRF One, but the HackRF is showing up in device manager, and is working with SDRTrunk and SDRSharp. I've tried the DFU batch script with the same result. I've tried using zadig to change the driver, but nothing.

eried, 2020-06-07:
check the video from here https://github.com/eried/portapack-mayhem/wiki/Update-firmware

If you still fail, record a 1 minute video with your attempt and describe exactly what you are doing and show how it fails

scriptitall, 2020-06-07:
[image]
[image]
[image]

Same results using admin cmd. I've watched and followed the video (tried 4 different cables are known to work with other devices)

scriptitall, 2020-06-07:
I have tried all the steps on all the USB ports on my computer with all 4 cables on each, but still no luck.

eried, 2020-06-07:
Check again the video. Install the driver included in the package replacing that one, skip the automatic driver detection it is just installing a generic usbhid.

scriptitall, 2020-06-07:
After some further investigation, windows defender was silently blocking the script from accessing devices.

Removing and reinstalling the drivers (from driver inside the firmware file) and fighting off defender has fixed it, but defender still warns me every time I try to run the update script.

scriptitall, 2020-06-07:
Also, if I can suggest a feature, it'd be great to be able to read log files on the portapack, instead of removing the sdcard

eried, 2020-06-07:
I cannot be Windows Defender, probably another antivirus solutions. Send the logs of what they say

scriptitall, 2020-06-08:
The logs are empty after a windows update (took 14 hours for windows to update and remove some of my programs...).

I disabled defender and it then worked. I also have AVG, which tested the scripts and allowed them, but defender kept warning me.

scriptitall, 2020-06-09:
[image]

I'm getting this error when I try to compile using docker

closed 2020-06-12 by scriptitall