portapack-mayhem/mayhem-firmware #56: Discord or Slack?

eried, 2020-06-07:
@johnelder @gregoryfenton @euquiq @nemanjan00 

What would be better for dev related polls and discussions? 

/polls "Discord" "Slack" "Facebook is ok" "Github issues are ok" "Other"

nemanjan00, 2020-06-07:
I vote for Slack, just because I already have client installed on phone and would like to avoid installing discord one due to privacy concerns

gregoryfenton, 2020-06-07:
I like Discord but many more people have facebook, however you have to consider that many features of the firmware are on sketchy legal ground and may be clamped down on at some point.

I also like discord as it can be used to sponsor content creators. "Want packet radio features? Pay us :P"

gregoryfenton, 2020-06-07:
Slack looks okay but it has limits such as 5GB max between all users, only one channel and no custom user groups (devs/support/whatever).
Ask ten people and you would get ten answers. To each their own.
Greg

eried, 2020-06-07:
Discord does not has such limits? I know in the free groups I use it is a pain because no uploads, etc

nemanjan00, 2020-06-07:
Gitter is the most used solution for this kind of thing...

https://gitter.im/

euquiq, 2020-06-07:
I am not using any of those tools, so I will jump-in on the one you decide.

theRFnoob, 2020-06-07:
I'd also vote for https://gitter.im/

It's open source and works well.

Disclaimer: I am not a dev, but I woudn't mind start having a look at the Havoc code with the help/guidance of others...

johnelder, 2020-06-07:
I've used discord for many things, but I'm willing to use other stuff as well. Never used slack, so no way to compare.

eried, 2020-06-08:
I also lean a bit for discord mainly because they have rich options and slack has that nasty limit of space. Gitter seems a bit too simplistic, in such a way that it will require addons and things to discuss stuff with fileshares, polls, screencasts to show an issue, etc

BorisDess, 2020-06-09:
Disord for me

euquiq, 2020-06-11:
DISCORD: GhostlyCrowd BorisDess eried gregoryfenton 
GITTER: theRFnoob nemanjan00 
SLACK: nemanjan00 

I think this "vote" recount is correct ?

johnelder, 2020-06-12:
You can add me to th discord vote

eried, 2020-06-13:
I think discord is the way then. Lets try for a test run: 

# Join: https://discord.gg/yrdKjGN

Ideally we should keep issues and discussions in Github, but more ludic and random conversations could be redirected to discord

closed 2020-08-10 by eried