portapack-mayhem/mayhem-firmware #59: Weird CLKIN signal

theRFnoob, 2020-06-07:
Hi, thanks a lot for starting this project. 
I am very excited to be able to interact directly with the developers :)

I have just installed the Mayhem firmware.
I have also produced a short video to help others with the installation process in Linux:
https://www.youtube.com/watch?v=0iZrGkfJN9o

As you can see at the end of the video (around min 9:00), I get a 10Mhz signal coming **out** of the CLKIN port, and I get nothing coming out of the CLKOUT port.

Is this a bug? It would be supergreat if we could have both:
- the possibility of using an external 10mhz clock reference (going to CLKIN), and
- also output the hackrf internal clock (from CLKOUT).

Thanks again. I am looking forward the announcement of a chat-room to discuss live with you guys.

eried, 2020-06-07:
I am very far from understanding the whole firmware, this is only an attempt to keep it clean and up to date in the releases :) so what you are asking is something I do not personally handle.

I will take a look tho, but the fastest answer would appear if you start checking the code. (I think everyone contributing here is slowly getting into it, so we dont have experts yet 😌)

theRFnoob, 2020-06-07:
Thanks, Eried. Of course! It's great effort to group all people currently interested in pushing further the project. Thank you for doing this :)

Maybe someone will see this Issue and will be able to fix it.

eried, 2020-06-08:
Hi again @theRFnoob, have you checked that the signal is not actually generated on the antenna sma? 

By reading the original hackrf documentation it seems the clkin and clkout are for Daisy-chaining another 10mhz reference because the files of the siggen haven't been touched in a long time

theRFnoob, 2020-06-08:
Hi @eried thanks for investigating.

You can see my setup (@9:33) here: https://youtu.be/0iZrGkfJN9o?t=573

No, the 10Mhz is not coming from the antenna. It is being emitted by the CLKIN port. 
Indeed when I try to check the CLKOUT port, I receive no signal.

Have you perhaps also tried with your Portapack and don't get this behavior?
That would be interesting indeed.

Also: I tried (not in the video) to go into HackRF mode, and then the 10mhz signal stops being outputed on CLKIN.

eried, 2020-06-10:
Ok. Because I am getting an error flashing and the error says some reference to the old hackrf.

GhostlyCrowd, 2020-06-11:
tat is very odd, i flashed this one i linked here to my hackrf one a few hours back. i used the latest build of hackrf tools located in https://github.com/pothosware/PothosCore/wiki/Downloads#windows-installer

GhostlyCrowd, 2020-06-11:
@eried 

```Microsoft Windows [Version 10.0.19640.1]
(c) 2020 Microsoft Corporation. All rights reserved.

E:\Users\jeffj\Desktop\HackRF_Tools>hackrf_info.exe
hackrf_info version: git-92ee144
libhackrf version: git-92ee144 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a32868dc295046cf
Board ID Number: 2 (HackRF One)
Firmware Version: local-a1a417d (API:1.03)
Part ID Number: 0xa000cb3c 0x00584761

E:\Users\jeffj\Desktop\HackRF_Tools>hackrf_cpldjtag.exe -x default.xsvf
File size 37629 bytes.
LED1/2/3 blinking means CPLD program success.
LED3/RED steady means error.
Wait message 'Write finished' or in case of LED3/RED steady, Power OFF/Disconnect the HackRF.
Write finished.
Please Power OFF/Disconnect the HackRF.

E:\Users\jeffj\Desktop\HackRF_Tools>```

eried, 2020-06-11:
You are completely right, I was using some old hackrf tools :)

```
C:\Users\erwin\OneDrive\Projects\HackRF>hackrf_cpldjtag -x default.xsvf
File size 37629 bytes.
LED1/2/3 blinking means CPLD program success.
LED3/RED steady means error.
Wait message 'Write finished' or in case of LED3/RED steady, Power OFF/Disconnect the HackRF.
Write finished.
Please Power OFF/Disconnect the HackRF.

C:\Users\erwin\OneDrive\Projects\HackRF>hackrf_info
hackrf_info version: git-92ee144
libhackrf version: git-92ee144 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc30349407
Board ID Number: 2 (HackRF One)
Firmware Version: local-7812144 (API:1.04)
Part ID Number: 0xa000cb3c 0x005c476b
```

Tomorrow I gonna test this with the scope. Sadly the environment changed from what @theRFnoob tested (new hackrf submodule/cpldjtag), but if I cannot replicate his "buggy output", then at least that's a positive.

matteo4plume, 2020-06-11:
Thanks for investigating guys!

@GhostlyCrowd even if you don't have a scope, you still should be able to use your HackRF rf-input port to "listen" to the CLKIN and CLKOUT ports :)

Anyway, I have just tried to update the CPLD successfully I believe (3 leds turning on and off). 
Now I have:

hackrf_info version: git-43e6f99
libhackrf version: git-43e6f99 (0.5)
Found HackRF
Index: 0
Serial number: .....
Board ID Number: 2 (HackRF One)
Firmware Version: local-c80ea84 (API:1.03)
Part ID Number: ....

And I still get the same problem document in my video once HackRF Portapack is running (mayhem):
- 10Mhz signal outputed on CLKIN
- nothing outputed on CLKOUT.

I am curious to see your tests too!

theRFnoob, 2020-06-11:
Thanks for investigating guys!

@GhostlyCrowd even if you don't have a scope, you still should be able to use your HackRF rf-input port to "listen" to the CLKIN and CLKOUT ports :)

Anyway, I have just tried to update the CPLD successfully I believe (3 leds turning on and off).
Now I have:

hackrf_info version: git-43e6f99
libhackrf version: git-43e6f99 (0.5)
Found HackRF
Index: 0
Serial number: .....
Board ID Number: 2 (HackRF One)
Firmware Version: local-c80ea84 (API:1.03)
Part ID Number: ....

And I still get the same problem document in my video once HackRF Portapack is running (mayhem):

10Mhz signal outputed on CLKIN
nothing outputed on CLKOUT.
I am curious to see your tests too!

eried, 2020-06-11:
So I got similar result everything updated to latest. On CLK OUT I get:
[image]

And on CLK IN:
[image]

Both around 10 mhz, but CLK OUT seems like just residual noise (it's in mV range). Strange. I need to do more tests, maybe without the portapack board, etc

theRFnoob, 2020-06-11:
Thanks @eried !
Ok so you confirmed my results (my analogue oscilloscope could not get the very small signal on CLKOUT, but I suspect that is just a coupled signal radiated from CLKIN).

In my tests, when running HackRF mode, the CLKIN signal disappears. So this seems (?) a mayhem related behaviour.

euquiq, 2020-06-12:
About updating the CPLD firmware, maybe it is a good idea to document such process in the wiki ?

theRFnoob, 2020-06-15:
Hello,

after discussions in the Mayhem discord chat, we realised that when the portapack board is on top of the HackRF, the clock being used is that of the PortaPack Board (usually higher precision). So it might not be so weird that on CLKIN we see that signal. I think the Portapack communicates the 10Mhz external signal using the P22 GPIO header (source: https://forums.hak5.org/topic/38290-gps-simulator/)

Regarding CLKOUT, I have noted that, going into hackRF mode and then executing: 

`hackrf_clock --clkout 1`

the CLKOUT port start to emit the 10mhz signal (of the HackRF clock, not of the Portapack clock).
If I then run: 

`hackrf_clock -r 3`

to obtain the infos regarding CLKOUT I get: 
`MS3:
	clock control = Up, Int Mode, PLL src A, MULTISYNTH SELF, 8 mA
	p1 = 9728
	p2 = 0
	p3 = 1
	Output (800Mhz PLL): 10.0000000000 Mhz
	output divider = 1`

The standard (i.e., no signal on CLKOUT) status is obtained again if I run  
`hackrf_clock --clkout 0`

which, after running 
`hackrf_clock -r 3`
again, gives:
`3
MS3:
	clock control = Down, Int Mode, PLL src A, XTAL, 2 mA
	p1 = 9728
	p2 = 0
	p3 = 1
	Output (800Mhz PLL): 10.0000000000 Mhz
	output divider = 1`

It would be ideal if:

1. we could activate this CLKOUT from within the Mayhem (for example, on the blue screen when we are in hackRF mode)
2. It would be even better if we could output on CLKOUT the clock of the Portapack clock (usually higher precision)
3. It would be perhaps useful in some occasions, if we could instruct the HackRF to *NOT* use the external (i.e., from Portapack) 10Mhz signal, and use instead its integrated clock.

eried, 2020-08-10:
I will close this issue now, but as a conclusion it would be nice if someone (i.e. @theRFnoob) created:
1) New individual requests for features (i.e. the CLKOUT control, etc)
2) Wrote some wiki text about the findings

closed 2020-08-10 by eried