portapack-mayhem/mayhem-firmware #67: I want to prettify all the .cpp/.hpp code

gregoryfenton, 2020-06-08:
**Is your feature request related to a problem? Please describe.**
My intention is to prettify the source code, remove excess indents and ensure brackets are aligned correctly.

**Describe the solution you'd like**
I will go through each .cpp/.hpp file using astyle and then by hand to ensure that there are no lines with just tabs, or lines with tabs at the end.

**Describe alternatives you've considered**
I have considered not doing this but I keep seeing little things that bug me :p

**Additional**
I would use astyle with the following options:
astyle -A1 -t -N -C -Y -p -xf filename

**Possible issues**
While doing this action no work should be done on the repositories and they should all  be synced beforehand and then immediately afterwards so any merge requests made afterwards will not fail.

**Timescale**
I envisage this taking no more than 30 minutes which would include pushing my changed files back to github.

gregoryfenton, 2020-06-08:
astyle -A1 -t -N -C -Y -p -xf < ui_sd_card_debug.cpp
/*
 * Copyright (C) 2016 Jared Boone, ShareBrained Technology, Inc.
 *
 * This file is part of PortaPack.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2, or (at your option)
 * any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; see the file COPYING.  If not, write to
 * the Free Software Foundation, Inc., 51 Franklin Street,
 * Boston, MA 02110-1301, USA.
 */

#include "ui_sd_card_debug.hpp"

#include "string_format.hpp"

#include "file.hpp"
#include "lfsr_random.hpp"

#include "ff.h"
#include "diskio.h"

#include "ch.h"
#include "hal.h"

class SDCardTestThread
{
        public:
                enum Result
                {
                        FailCompare = -8,
                        FailReadIncomplete = -7,
                        FailWriteIncomplete = -6,
                        FailAbort = -5,
                        FailFileOpenRead = -4,
                        FailFileOpenWrite = -3,
                        FailHeap = -2,
                        FailThread = -1,
                        Incomplete = 0,
                        OK = 1,
                };
                std::string ResultStr[10] =
                {
                        "Compare",
                        "Read incomplete",
                        "Write incomplete",
                        "Abort",
                        "File Open Read",
                        "File Open Write",
                        "Heap",
                        "Thread",
                        "Incomplete",
                        "OK",
                };

                struct Stats
                {
                        halrtcnt_t write_duration_min { 0 };
                        halrtcnt_t write_duration_max { 0 };
                        halrtcnt_t write_test_duration { 0 };
[... 426 lines cut ...]
                        text_test_write_rate_value.set(
                            format_bytes_per_ticks_as_mib(stats.write_bytes, stats.write_duration_min * stats.write_count) + " " +
                            format_bytes_per_ticks_as_mib(stats.write_bytes, stats.write_test_duration)
                        );

                        const auto read_duration_avg = stats.read_test_duration / stats.read_count;

                        text_test_read_time_value.set(
                            format_ticks_as_ms(stats.read_duration_min) + "/" +
                            format_ticks_as_ms(read_duration_avg) + "/" +
                            format_ticks_as_ms(stats.read_duration_max)
                        );

                        text_test_read_rate_value.set(
                            format_bytes_per_ticks_as_mib(stats.read_bytes, stats.read_duration_min * stats.read_count) + " " +
                            format_bytes_per_ticks_as_mib(stats.read_bytes, stats.read_test_duration)
                        );
                }
                else
                {
                        text_test_write_time_value.set("Fail: " + thread.ResultStr[thread.result() + 8]);
                }
        }

} /* namespace ui */

gregoryfenton, 2020-06-08:
`astyle -A1 -t -N -C -Y -p -xf < ui_sd_card_debug.cpp
/*
 * Copyright (C) 2016 Jared Boone, ShareBrained Technology, Inc.
 *
 * This file is part of PortaPack.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2, or (at your option)
 * any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; see the file COPYING.  If not, write to
 * the Free Software Foundation, Inc., 51 Franklin Street,
 * Boston, MA 02110-1301, USA.
 */

#include "ui_sd_card_debug.hpp"

#include "string_format.hpp"

#include "file.hpp"
#include "lfsr_random.hpp"

#include "ff.h"
#include "diskio.h"

#include "ch.h"
#include "hal.h"

class SDCardTestThread
{
        public:
                enum Result
                {
                        FailCompare = -8,
                        FailReadIncomplete = -7,
                        FailWriteIncomplete = -6,
                        FailAbort = -5,
                        FailFileOpenRead = -4,
                        FailFileOpenWrite = -3,
                        FailHeap = -2,
                        FailThread = -1,
                        Incomplete = 0,
                        OK = 1,
                };
                std::string ResultStr[10] =
                {
                        "Compare",
                        "Read incomplete",
                        "Write incomplete",
                        "Abort",
                        "File Open Read",
                        "File Open Write",
                        "Heap",
                        "Thread",
                        "Incomplete",
                        "OK",
                };

                struct Stats
                {
                        halrtcnt_t write_duration_min { 0 };
                        halrtcnt_t write_duration_max { 0 };
                        halrtcnt_t write_test_duration { 0 };
[... 426 lines cut ...]
                        text_test_write_rate_value.set(
                            format_bytes_per_ticks_as_mib(stats.write_bytes, stats.write_duration_min * stats.write_count) + " " +
                            format_bytes_per_ticks_as_mib(stats.write_bytes, stats.write_test_duration)
                        );

                        const auto read_duration_avg = stats.read_test_duration / stats.read_count;

                        text_test_read_time_value.set(
                            format_ticks_as_ms(stats.read_duration_min) + "/" +
                            format_ticks_as_ms(read_duration_avg) + "/" +
                            format_ticks_as_ms(stats.read_duration_max)
                        );

                        text_test_read_rate_value.set(
                            format_bytes_per_ticks_as_mib(stats.read_bytes, stats.read_duration_min * stats.read_count) + " " +
                            format_bytes_per_ticks_as_mib(stats.read_bytes, stats.read_test_duration)
                        );
                }
                else
                {
                        text_test_write_time_value.set("Fail: " + thread.ResultStr[thread.result() + 8]);
                }
        }

} /* namespace ui */`

eried, 2020-06-08:
Good! it is that same style as Visual Studio code uses by default? 

Also, there is a lot of orphan files and such but I have not found a nice tool to deal with them

gregoryfenton, 2020-06-08:
* I have just done a test run and then built the image, the image is identical to my before attempt.
* I don't use Visual Studio, I use good  old gnu nano :)
* Please give me an example of an orphan file and I'll see what I can do :)

eried, 2020-06-08:
Visual Studio Code works awesome with the code without any config, you can see references, etc

Anyways, I will check it in VSCode later, for the orphan files, 2 examples: ui_handwrite.cpp, ui_playdead.cpp

marchelh, 2020-07-08:
I also want to add, that all the compiler warning must actually be fixed.

gregoryfenton, 2020-07-08:
I totally agree with that @marchelh, finding the one error line in the midst of 500 warning lines is a pain.

eried, 2020-07-08:
Yes I agree too. I tried to remove few but it is a long way to go. First I will do what I told you @gregoryfenton about the "reindentation bot"

eried, 2020-08-10:
This will be closed for now, but the idea of using the bot as the indentation "author" is still present on my mind. I wish there was a cleaner way to keep the collaboration history...

closed 2020-08-10 by eried