portapack-mayhem/mayhem-firmware #69: Pressing the screen to select a menu item inconsistency

gregoryfenton, 2020-06-08:
**Describe the bug**
Entering a sub-menu item and returning to the sub-menu differs whether you use the jog wheel or click on the button manually

**To Reproduce**
Steps to reproduce the behavior:
1. Open a menu with a sub-menu such as Debug using the jog wheel.
2. Open an item on the sub-menu such as Temperature
3. Return to the Debug menu by clicking Done
* Note at this point that you are back in the Debug menu with Temperature highlighted.

4. Exit the Debug menu.
Instead of the jog wheel, this time use the touchscreen:
5. Press on Debug
6. Press on Temperature.
7. Press on Done to return to the Debug menu.
* Note at this point that you are back in the Debug menu with Memory highlighted.

**Expected behavior**
5 to 7 above should have the same result as 1 to 3 - Temperature should be highlighted.
Pressing on a button should tell the main UI the selection has changed as if the jog wheel moved.

**Affected versions**
All since button menus were introduced, as far as I can tell this has never been noticed previously.

**Additional**

eried, 2020-06-08:
I see, maybe is related with not providing the default button

euquiq, 2020-06-08:
I have no such problem. For me it works fine ... but I have the ".." button enabled, which I assume is the way the menu was designed on the first time

eried, 2020-06-08:
I checked old havoc and it is the same behaviour. @euquiq you have the same problem, it is even on very old havoc firmwares (any version that implemented the buttons instead of the scroll list)

euquiq, 2020-06-09:
With my ".." button enabled on each menu, my portapack works just as @gregoryfenton expects, instead of what he gets.

eried, 2020-06-09:
Highlight Receive with the buttons, tap with your finger transmit, tap back with your finger. Which one is highlighted now?

euquiq, 2020-06-11:
Yep, AFAIK / see, the buttons grid menu behaves as expected when using the cursor keys and rotary wheel. 

The on_key() and on_wheel() are overriden and locally coded inside /firmware/application/ui/ui_btngrid.hpp (and the code dealing with these events is clearly placed inside the .cpp file)

But the touchscreen event on_touch() goes  into a different path, the button itself executes the  on_Select(function), again AFAIK / SEE, this is in general, buttons behave like this.

I am at the limit of my C++ knowledge, learning a bit more every day. If anyone can help with this, maybe we can either fix this bug, or decide that it is inherent to the way the buttons are dealt.

Also maybe I am 100% wrong. I am still experimenting and trying alternatives on this issue. It definitely doesn't help that between each code / compilation, I got looong minutes of pause while compiler churns away.

eried, 2020-06-11:
I don't know if is really a bug. It can be considered a feature (i.e. you put the cursor on replay, tap record, record something and then tap back and enter gets you to replay)

euquiq, 2020-06-12:
> I don't know if is really a bug. It can be considered a feature (i.e. you put the cursor on replay, tap record, record something and then tap back and enter gets you to replay)

 I managed to "fix" the bug / feature, for what it is worth :) Now we need to determine if we add the fix into the repository, and buttons behave just as cursor / wheel ... or not. (Cry / laugh) (2 days of hair pulling -but fun- test / understand / retest) jajaa ... 

Let me explain the fix:

The whole buttons behavior for the menues is defined in `/firmware/common/ui_widget.cpp`  on the **NewButton** widget. This includes even launching the function that the menu item calls inside (with **on_touch()** and then **on_Select()**).

As I explained in my earlier post,  keys and wheel navigation, on the other hand is handled inside `/firmware/application/ui/ui_btngrid.hpp`, including the actual mechanism for button "memory" ... which actually seems to be the right place for this.

I decided to move the on_touch() processing, from **ui_widget.cpp** into **ui_btngrid.hpp** where it should belong.

So first I commented out the whole `bool NewButton::on_touch(const TouchEvent event)` function in **ui_widget.cpp**.

Secondly, I moved the  declaration `on_touch(const TouchEvent event) override;`  from **ui_widget.hpp** into  **ui_btngrid.hpp** (the place where it made sense to process  the button on_touch()) ...

And in third place, I added the the following "revamped" on_touch() function inside **ui_btngrid.cpp**:

```
bool BtnGridView::on_touch(const TouchEvent event) {
	size_t i = 0;
	for(const auto child : children_) {
		if( child->screen_rect().contains(event.point) ) {
			switch(event.type) {
			case TouchEvent::Type::Start:
				return set_highlighted(i-1);
			case TouchEvent::Type::End:
				if( menu_items[highlighted_item].on_select ) 
					menu_items[highlighted_item].on_select();
				return true;

			default:
				return false;
			}
		}
		i++;
	}
	return false;
}
```

And "voilá", the on_touch is also processed on the ui_BtnGrid context, with the same mechanisms as wheel and key navigation.

BTW: the "newbutton" widget inside ui_widget.cpp was there only to be used by ui_BtnGrid, so moving it entirely out of there and relocating it inside ui_BtnGrid should be no biggy (as far as I can tell).

On a side-note, I found a touchscreen sensitivity threshold parameter, which also may be of serious use, in the event of a low quality chinese touchscreen... see inside `/firmware/application/touch.hpp`  Line 215.

eried, 2020-06-12:
Great finding! 🤓 I wonder if that will affect usage of other screens since is actually moving focus based on touch now (like how freqman feels while operating, etc)

euquiq, 2020-06-12:
Nope! it does not affect anything, since this override is only for the menues .. and the NewButton is used only in the menues. I will send you the PR, so you can inspect it

euquiq, 2020-06-12:
Anyway, AFAIK I may be misunderstandig the "override" specifier, but the original `virtual bool on_touch(const TouchEvent event);` is actually declared right there, inside ui_widget.hpp.

Then in there too there are several `bool on_touch(const TouchEvent event) override;` (one for each widget accepting touch).

One of those widgets if the NewButton, which is ONLY used for the menues.

Since **NewButton** widget is **only used for the menues**, when I comment out (or delete) that on_touch() override, because I will be processing the on_touch() inside ui_btngrid.cpp, no harm is done at all, since all other on_touch() overrides will still be in effect for the rest of each other widgets.

I am testing my copy of your current version, with these small changes, and testing each button type using on_touch(). After I verify everything is just as expected, I will send you the PR.

euquiq, 2020-06-12:
Again, I tried to do a clean PR, but I think that Github added other stuff ....

euquiq, 2020-06-12:
The clean PR was done in the following way:

`git remote -v `
 (checking your master branch is upstream)

`git checkout -b fix-touch-menu-history upstream/master  `
(creating a new branch from my already up-to-date copy of your main branch)

`git cherry-pick 4cf4b6ef4e9bab9dd2360cf380bde22963a59ae8`
(inserting the specific commit which included all necessary code changes in the new branch)

`git push -u origin fix-touch-menu-history`
(pushing it into github)

Then I went into your repository, and there it was ready to be PR'ed.

eried, 2020-06-12:
Few days ago I recorded a video how to do a clean PR, let me see if is good enough to upload it. I have never used cherrypick.

euquiq, 2020-06-15:
I did close my PR regarding this bug.  See my last comment (when closing) in:

https://github.com/eried/portapack-mayhem/pull/76

It remains to be seen if this is really a BUG, or an ENHANCEMENT ... or actually leaving it as it is now, allows for dual quick selection: You can place your highlight on button REPLAY, but press with touchscreen on RECORD. So you record, go back, and just press the select button to get into REPLAY.

eried, 2020-08-10:
I am going to close this because at the end was not clear if it was actually a bug. If this is not the case please feel free to reopen the issue.

closed 2020-08-10 by eried