portapack-mayhem/mayhem-firmware #70: Update ADS-B airlines database

sliceratwork, 2020-06-08:
Hey @eried 

Here's a list of airlines taken from [Wikipedia](https://en.wikipedia.org/wiki/List_of_airline_codes_(A)).

There should be a lot more than in the original file. 

Thanks.

eried, 2020-06-08:
Cool, this will probably be included on next revision. If you want to create the file and share it even better 🤓 (it only requires python, I think)

gregoryfenton, 2020-06-08:
I have merged the data from both sources but I can't find the original airlines.txt to recreate the structure. Until then, here is what I have: [greg_airlines.txt](https://github.com/eried/portapack-mayhem/files/4749018/greg_airlines.txt)

eried, 2020-06-08:
Which both sources? 

The original structure is:

`3LETTERCODE, Name (Country)`

However there are multicountry airlines, which I am not sure how to deal with. Also that list has inconsistent naming, for example Russia vs Russian federation

gregoryfenton, 2020-06-08:
I used  the list of airlines in the first comment, and stripped the data from the wikipedia page.
I then filtered for duplicate lines.

I'm off to bed now, but if the data is as simple as
3LC,Name (Country)
I'll sort it first thing tomorrow.

I'm guessing a duplicate 3LC would just pick the first in the list which may not be ideal but it's better than nothing 👍

eried, 2020-06-08:
Or maybe combine them in 1, @sliceratwork how do they appear on the screen? How many chars for the airline and country are visible?

sliceratwork, 2020-06-09:
I realized I din't link to the original file I mentioned. Based on the [original Wiki](https://github.com/furrtek/portapack-havoc/wiki/ADS-B-receiver),  download [AcarSDeco2](https://drive.google.com/open?id=1iBz6GrBSWrBdA68o3fD0YpQXpwMnXYNK) and extract datasets/airlines.txt.

@gregoryfenton maybe you can use your gnu nano magic again and merge everything into one file :)

@eried unfortunately I don't have my portapack with me now. Will check at the end of the week.

gregoryfenton, 2020-06-09:
This is the latest version of airlines.txt, merged and formatted as required.
[airlines.txt](https://github.com/eried/portapack-mayhem/files/4752360/airlines.txt)

eried, 2020-06-09:
What do we do with those with more than 1 comma like `ACR,Aerocenter, Escuela de Formación de Pilotos Privados de Avión(Spain)` and special tildes?

I have not checked the code that works around them so I cannot tell.

ALSO: there is a space after the first comma and before the last open parentesis (in the original parser tool)

gregoryfenton, 2020-06-09:
[airlines.txt](https://github.com/eried/portapack-mayhem/files/4752494/airlines.txt)

eried, 2020-06-12:
@sliceratwork have you tried making the new .db with @gregoryfenton 's file? If it works ok we can update the built-in file for everyone

sliceratwork, 2020-06-13:
@eried will have a go tomorrow morning and let you know.

Thanks for keeping Havoc alive.

sliceratwork, 2020-06-16:
@eried I've created a new airlines.db from the text file and every plane comes up as unknown.

I don't think there's a problem with the text file itself, but with the python script. I had a quick look at it and it seems it's expecting the data in a slightly different format.

Here's how my [airlines.db.txt](https://github.com/eried/portapack-mayhem/files/4785376/airlines.db.txt) file came out. Compare it to the one in the sdcard folder and you'll see the differences.

Any thoughts?

eried, 2020-06-16:
Yes I see, the original .db has the 3 letter codes at the beginning and yours in between of the airline names. Maybe the real script was never updated and another one was used for creating the db. I am not sure if there is another script around, but the alternative is to do reverse engineering on the code reading the file to recreate the script. 

What I can quickly see is:
https://github.com/eried/portapack-mayhem/blob/master/firmware/application/apps/ui_adsb_rx.cpp#L136

This is the routine to read the name. It looks for the code at the beginnning and then jumps to the name, that is fixed length (so we probably need to adjust that also in @gregoryfenton's file if is not already handled by the script)

If you want to give it a go to modify the .py script, go ahead. I will eventually check it, but not immediately

sliceratwork, 2020-06-16:
Thanks for having  look.

I'll mess around with the python script tonight and see what I can come up with.

eried, 2020-06-16:
Cool thanks! something similar happened with the bitmaps, I was fixing that script and it was actually not the right one so it is a very probably possibility that the current script is simply an old unnused one. Just share your findings here

eried, 2020-06-16:
Actually maybe @gregoryfenton and @euquiq have some inputs. Maybe we could just change the way they are stored and read:

https://github.com/eried/portapack-mayhem/blob/master/firmware/application/apps/ui_adsb_rx.cpp#L136

1) this scans 0x0000~0x2000 for the code, that is 3 bytes of the code + null
2) reads padded data for the name based on the position 

What I think would be more flexible (because the current implementation has a limiting factor) is just to have:
`3 bytes, 1 null, right padded x bytes of the name, 1 null, `

We have a discord (link on the main readme) to discuss

abvgdeyoj, 2021-04-18:
> Here's [a list of airlines](https://github.com/eried/portapack-mayhem/files/4747359/airlines.csv.txt) taken from [Wikipedia](https://en.wikipedia.org/wiki/List_of_airline_codes_(A)). 

Please remember that the list linked is under [CC BY-SA 3.0](https://en.wikipedia.org/wiki/Wikipedia:Text_of_Creative_Commons_Attribution-ShareAlike_3.0_Unported_License), while the project is [GPLv2](https://github.com/eried/portapack-mayhem/blob/next/LICENSE).
Embedding this list into the project may be [not legally possible](https://opensource.stackexchange.com/questions/2212/is-stack-exchanges-cc-by-sa-v3-0-content-compatible-with-the-gpl).

eried, 2021-04-19:
Maybe the SDCARD file contents can include multiple licenses for the content? (they are already a separate entity)

ArjanOnwezen, 2021-11-24:
Solved with #434

closed 2021-11-24 by ArjanOnwezen

joyel24, 2023-05-26:
@ArjanOnwezen Why don't you claim your 10$ bounty ? https://app.bountysource.com/issues/92194917-update-ads-b-airlines-database

ArjanOnwezen, 2023-05-26:
@joyel24 I just like to contribute. But that said, if the bounty reaches 6 figures, I'll probably grab it.. ;)

joyel24, 2023-05-26:
> @joyel24 I just like to contribute. But that said, if the bounty reaches 6 figures, I'll probably grab it.. ;)

In my opinion, it's better to take the bounty as the amount has been paid by some people...

They pay for the dev who will be able to implement things they ask. But if the dev don't claim bounties, the money will go to the enterprise... Sad 😭