portapack-mayhem/mayhem-firmware #73: GSM scanner

marcoSanti, 2020-06-11:
Could it be possible to implement a gsm scanner for bts signals (like what the grgsm_scanner from the gs gsm suite, does) to get a list of all BTS stations available. It would be nice if it could work for generations from 1G to 4G LTE. 
What this app should do is to show a list (ordered by tx channel) of all the transmitting BTS in the current zone.

eried, 2020-06-11:
Can you show how do you do it with that scanner?

marcoSanti, 2020-06-12:
Yes, so basically you launch the app and you get something like this:
[image]
It would be nice to have shown as a list the ARFCN field and the rx power (and by clicking on on each of the entry on the list, to show the other fields).
The code is from this repo (https://github.com/ptrkrysik/gr-gsm/)
Unfortunately I have zero experience in programming in c++ and I am a total noob in python, so porting feature is still impossible for myself (i have knowledge in many other languages, but not thoose ones), otherwise i would have tried for myself. Hope to have explained myself, Tku

eried, 2020-06-12:
I dont think anyone here is/considers himself an expert in C++ besides what we were forced on Uni (lol). Sounds like a nice app to build but there is a lot of bugs and small inconsistencies to clean still so I dont see this being attempted by anyone but [you](https://github.com/eried/portapack-mayhem/wiki/How-to-collaborate#coding-new-stuff-or-fixing-bugs) 🤣 

[image]

have you tried it with the hackrf? or just bladerf?

marcoSanti, 2020-06-12:
I have not tried it with bladerf since i do not have one! the photo i posted is from a video i found: unfortunately this morning my pc updated anth grgsm sopped working... I am still trying to fix it..
Btw the scanner app works perfectly with the hackrf since it only listens to the bts. you need a bladerf if you want to create a rogue bts, which i won't suggest doing since it is absolutely illegal (at least in Italy)...

smannchen, 2020-08-04:
GSM has just about been completely phased out in the US. LTE cell scanner seems more promising, however I doubt it will ever be implemented in the portapack. Here's a link to the software for pc though. https://github.com/JiaoXianjun/LTE-Cell-Scanner

marcoSanti, 2020-08-04:
Yeah i know, but phones should be supporting it for a while... in any case i do not think I will be able to port the code you linked into portapack😅

closed 2023-12-04 by eried

funder7, 2024-09-24:
Hello, I might try to do this if you're still interested.
I never used both tools, but it's something that I'm interested to use in the next future, and it would be great to have such features in a portable device. 

At first glance, `grgsm_scanner` is written in python, so I guess it cannot be used as-is. 
`LTE-Cell-Scanner` on the other hand, is written in C++ and already supporting the HackRF, so a port should be easier. 

2G and 3G Networks are still in use here in Europe, so I wouldn't exclude them (or consider 3G at least).

Let me know!