portapack-mayhem/mayhem-firmware #75: docker compilation errors

JonaldJohnston, 2020-06-12:
when running `docker run -it -v ${PWD}:/havoc portapackccache`, i keep getting the following error immediately:
```
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /opt/build/armbin/arm-none-eabi
CMake Error at CMakeLists.txt:48 (add_subdirectory):
  add_subdirectory given source "hackrf/firmware/hackrf_usb" which is not an
  existing directory.

-- Configuring incomplete, errors occurred!
See also "/havoc/build/CMakeFiles/CMakeOutput.log".
```
aaaa what do i do?

eried, 2020-06-12:
This is super strange, can you detail more how you created it and your environment etc?

JonaldJohnston, 2020-06-15:
compiling on linux mint 19.3, installed docker and docker.io through apt,
followed instructions on the wiki, running `docker build -t portapackccache -f dockerfile-nogit .`, `mkdir build`, and `docker run -it -v ${PWD}:/havoc portapackccache`

eried, 2020-06-15:
Ah, probably you fetched the git repo via command line, you should also sync the submodules.

eried, 2020-06-15:
Something like: https://stackoverflow.com/questions/1030169/easy-way-to-pull-latest-of-all-git-submodules

Github Desktop does it all automatically

JonaldJohnston, 2020-06-17:
that seems to have worked! thanks

closed 2020-06-17 by JonaldJohnston