portapack-mayhem/mayhem-firmware #77: Whip antenna length calculator give wrong values

euquiq, 2020-06-13:
I was taking a peek into https://github.com/eried/portapack-mayhem/issues/48 when I realized that the length returned by the antenna length calculator is off ... sometimes way off!

i.e.: Consider an antenna for frequency = 2400Mhz (2.400.000 hz)

(as used in **ui_whipcalc**) metric: Speed of light: 299792458 m/s

speed / frequency = 0,1249m (12,49cm full wave) or 3,1cm quarter wave

In this case, the metric result is ok but in imperial it is wrong:

Full wave: If I convert 12,49cm into inches, google brings me: 4.91 inches but antenna length calc gives 4.0in ! On Quarter Wave: metric 3.1cm to inches is about 1.22in ... but whip calc gives 1.0in

There is a flaw in the way a function inside ui_whipcalc (**to_string_dec_int()**) returns the value  of a number as a string: Decimals are not rounded up which seems to end up returning values which are way off.

This function is used in another parts of the code !

Now I am focusing on fixing the Antenna calculator (as a start).

eried, 2020-06-13:
I see you are totally correct @euquiq, the conversion is wrong. I have not poked into the calculator because I am waiting for a nanovna to actually measure the cheap whip antenna included in my kit. My idea is to write a interpolator with the exact values. I do not have an ANT500 nor an ANT700, but if I document the measurement procedure some other people will do the measurement. 

In any case if you fix the conversion, this will be helpful!

Short story:
I want to have a real measurement and interpolation for:
1) Genuine ANT500
2) Genuine ANT700
3) Included cheap antenna

That is why I ordered that nanovna 2 months ago, but with covid china is working slower than normal.

Dirk275, 2020-06-13:
I'm lucky to have ANT500 and NanoVNA.
How can I help?

euquiq, 2020-06-13:
Hi @Dirk275 ! As @eried commented on issue https://github.com/eried/portapack-mayhem/issues/48 perhaps you could extend your NanoVNA antenna give us the measure for each segment ? 

If it is not too much, perhaps you can do the same with ANT500, altough it is already coded, I would very much want to verify the actual values against the coded calculation.

Thanks!

Dirk275, 2020-06-13:
Ok, I'll do it tomorrow.
Do you need only the lengths of each element and/or the resonance freq measured with NanoVNA?

euquiq, 2020-06-13:
ah! sorry now I understand. I am ashamed hehe to say that I confused the name **NanoVNA** with an **antenna model** (facepalm). I see, now, the NanoVNA is similar to a vector antenna analyzer (I got a low cost PS100). 

Then only the second part of my comment is valid :) maybe you can measure each segment as confirmation ?

Did you already try to compare extending your ANT500 with the metric numbers that Portapack antenna calculator gives for a particular frequency (say 433mhz) with your NanoVNA ? 

As The RF Noob did in here: https://youtu.be/MzbAEBghcPM

Dirk275, 2020-06-13:
Yes, I can measure both antennas 1 and 3 of eried's post above and get their resonance frequency for each element. Then we can compare with the Portapack values.

eried, 2020-06-13:
> Ok, I'll do it tomorrow.
> Do you need only the lengths of each element and/or the resonance freq measured with NanoVNA?

That would be perfect! length vs resonance! thanks a lot

Strangely in that video of "The RF Noob" that antenna looks different to my cheapo one

Dirk275, 2020-06-13:
Yes maybe the antennas delivered with not original HackRF or HackRF Portapack are not the same or the same size.
My "cheap antenna" that came with HackRF Portapack has these lengths:
Min. 118 mm, max. 476 mm, measured from top to the white plastic ring above the SMA plug.

eried, 2020-06-13:
Yeah, mine is that same size, maybe it is the "new H2 batch". I do not think china actually produces many different versions if they already achieved minimum price

[image]

Dirk275, 2020-06-13:
... seems to be identical.
Then I'll measure ANT500 and this antenna tomorrow.

eried, 2020-06-13:
> ... seems to be identical.
> Then I'll measure ANT500 and this antenna tomorrow.

By the way, mine felt so flimsy... I was almost going to buy another one (or design a TPU sleeve for it) however a drop of super glue on the screw and proper tightening fixed everything! JFYI

Dirk275, 2020-06-14:
So, theory and practical measurements ... 
1st the "cheap ant" (118..476 mm):
Elements--Length--freq of lowest S11 (lambda)
1--118mm--162MHz (1/16)
2--183mm--418MHz (1/4)
3--253mm--303MHz (1/4)
4--326mm--118MHz (1/8)
5--399mm--96MHz (1/8)
6--476mm--80MHz (1/8), 644MHz (1/1)

2nd the ANT500 (185..862mm):
Elements--Length--freq of lowest S11 (lambda)
1--185mm--413MHz (1/4)
2--315mm--244MHz (1/4), 490MHz (1/2)
3--450mm--340MHz (1/2)
4--586mm--262MHz (1/2)
5--724mm--424MHz (1/1)
6--862mm--355MHz (1/1)

I have measured reflection with my NanoVNA from 50 kHz to 900 MHz and checked for the frequency of lowest S11 peak. Due to the max. frequency of 900 MHz I cannot see low S11 peaks above that limit, but they will surely exist.

Dirk275, 2020-06-15:
I didn't look at the code yet, but ...
After my measurements it is clear, that there is a deviation between the measured and the calculated results.
The reason is obviously, that the calculation doesn't take account of the NVP.
Because the antennas are not ideal, they are functionally shorter than they are physically.
For example I found for the ANT500, that the shortest length 185 mm is functioning as a 181 mm antenna. So the NVP should be sth. like 0.98 for the ANT500.
If we would adapt the antenna calculator, then better results would be reached by adding the NVP to the calculation.
If you would plan to do so, I would have to make more measurements to calculate the NVP of the used antennas (each has it's own depending on the material/diemensions ...).
... or would this all overshoot ...?

eried, 2020-06-15:
I think extra information is never enough, I hope I can corroborate your measurements when I get my nanovna. Meanwhile is not implemented to the app, just having it here or if you think is wiki-worthy material: on the wiki too https://github.com/eried/portapack-mayhem/wiki/antennas

euquiq, 2020-06-16:
I would say this issue is closed, since the examples of bad calculation I give on the issue, are solved with the patch merging https://github.com/eried/portapack-mayhem/pull/78 .

closed 2020-06-16 by euquiq