portapack-mayhem/mayhem-firmware #79: Bias Voltage not working?

theRFnoob, 2020-06-14:
When I go (using the Mayhem firmware, of course) to 

`Settings--> Radio
`
and click on

`"Turn Bias Voltage = ON"
`
I don't seem to get any bias voltage on the RF output of the HackRF.
Also, if I go to (for example) Receivers--> Audio and start listening, I don't get any bias voltage.

Furthermore, when I reset the hackrf, the setting "Turn Bias Voltage ON" is forgotten, and the unit always restarts with 

`"Turn Bias Voltage = OFF"
`

In other words, for me the Bias Voltage funcitonality doesn't seem to work at all.

eried, 2020-06-14:
It is the same with the Bias icon on the upper status bar? (to the right of the "Sleep" button)

theRFnoob, 2020-06-14:
@eried yes, also turning ON/OFF the Bias icon has no effect.

NOTE: the bias icon is correctly linked with the Settings-Radio-"Enalbe Bias", in the sense that they show the same ON/OFF status. But, again, I get no DC on the line.

eried, 2020-06-14:
Since you have a lot of equipment could you test this and the CLK OUT thing with GridRF version http://gridrf.com/products/detail/id/11.html, some old Havoc and possibly original PortaPack firmware (at least the CLK OUT thing)?

We can dig into the code if somebody has some actual clue when this changed, or maybe it never worked and it has to be removed or documented.

theRFnoob, 2020-06-14:
I just tried with an older Havoc version (the same present in the .zip file of the Mayhem distribution).

The BIAS VOLTAGE doesn't seem to work there, too.

I don't mind testing several firmwares, but I am not so familiar with them, their webpages, etc.
Would you mind listing here a list of links to the .bin files of all the firmwares you want me to test?

Thanks!

eried, 2020-06-14:
Yes, I think it is useful for any testing so I uploaded them here:
https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

theRFnoob, 2020-06-14:
Thanks, 
so I tried all these firmwares.

OLD_2018_sharebrained_portapack-h1-firmware.bin --> "nightly-1fe2d7b"
OLD_2019_gridrf_portapack-havoc-gps.bin --> "Git Hash 30db228"
OLD_2019_portapack-h1-havoc.bin --> "Git hash 30db228" (so unless I made some mistakes, this is equal to the one above, even if the bins have different size. Maybe they differ only on some additional applications installed.)
OLD_2020_portapack-h1-havoc.bin --> "Version local-f4b740d2"

I can confirm that in **all** of them:

1. CLKIN emits a 10Mhz signal
2. CLOUT does not emit anything.
3. The RF output has a DC bias of 1.6V but when you apply a 50ohm load to it it drops to 30mV
4. The "BIAS ON/OFF" does not work, i.e., it seems to do nothing at all.

So I guess the CLKIN/CLKOUT never really worked properly, and the BIAS voltage also never really work (at least with my hardware).

eried, 2020-06-14:
How do you test the bias? Just directly with a multimeter in ANT out?

theRFnoob, 2020-06-14:
I did use an oscilloscope:
- when connected directly (using a coax cable) it shows 1.6V
- when 50ohm terminated, it shows 30mV

theRFnoob, 2020-06-14:
I just added this quick video to dropbox to show the measurement:
https://www.dropbox.com/s/diudy8o9num9s3m/2020-06-14%2013-45-06.mkv

I have to correct myself: 
- when the ANT-out port is 50ohm terminated, I see about 3mV not 30mV as stated above. Anyway, close to zero.
- when the ANT-out port is not terminated (or actually, terminated by 1Mohm of my scope) I see 1.6V
- when I connect the ANT-out port to a miniwhip antenna, I see about 0.037V = 37mV.

eried, 2020-06-14:
BTW, a lot of old versions have the same name in version, it is just from the build process. That was one of the main confusions back then when people were trying to find "latest". 

In the same link I added a 2017 version that had the nearest commit saying BIAS was actually working, could you check that as a last test?

theRFnoob, 2020-06-14:
I just checked the 2017 version. It doesn't work for me either, unfortunately.

I still get the 1.6V (in theory it should be 3.3) and nothing changes if I turn BIAS=ON or BIAS=OFF.

Perhaps it's just my hardware not working. Maybe I have damaged it in some experiments.
It would be interesting to have others (with a multimeter or an oscilloscope) test it as well.

euquiq, 2020-06-14:
Let me try on my own H1: I measure the antenna connector directly with a tester.  with the bias OFF  I get about 4mv ... bias ON the same, the tester readout doesn´t move, still shows the 4mv.

eried, 2020-06-15:
So it might be that the chinese copies saved on parts? OR it was never implemented. Any of those will require more research. It would be nice if somebody with a genuine "kit" could test.

theRFnoob, 2020-06-15:
Or does someone know how to activate the bias via software using the HackRF-mode? If it works for the HackRF, it's not a hardware problem. 

I tried to search a bit but couldn't find a simple command to activate the bias via command line in linux.

eried, 2020-06-15:
In the "new" tools I saw a parameter to do it;
`hackrf_transfer -p 1`

I am checking if the new hackrf module will break the portapack. There is also commands for controlling the UI (that blue display that appears in portapack) but I have not tried any of those.

theRFnoob, 2020-06-15:
Ok, I have just tested the 
`hackrf_transfer -p 1 -r ./data.rec`

command. Unfortunately **it does not work** with or without the portapack module on top of the Hackrf, in the sense that the command runs, but no bias voltage appears.

It might be the case that, indeed, the chinese vendors saved some money and didn't include the BiasT. Although this seems a bit bizarre (it probably costs more time to them to modify/redesign the board than to add the few capacitors needed for the bias-T circuit).

It only remains to check if this functionality works on "original" HackRF. But I suspect there are very few of them around.

theRFnoob, 2020-06-15:
i suspect there might be something wrong with the HackRF (not portapack) firmware I am running at the moment.

Indeed I am only getting a VEEERY (about 1mv ppk) weak signal on the CLKOUT port (10mhz). My counter cannot even pick it up. I am sure, a couple of months ago, it used to output a clear 10Mhz signal.

Nothing comes out of the CLKIN port.

Dirk275, 2020-06-17:
For me the Bias-T function is ok. See pictures:
[image]
[image]

Perhaps your Bias-T output is broken:
1. If you connect a 50 Ohm resistor to the antenna input (see your post 3days ago), then this draws 66 mA. The HackRF output is able to drive 50 mA max. at 3.3 V, that's the limit, so 66 mA might still work without destroying the HackRF but perhaps might not.
2. If you connect a Mini Whip antenna base board (without it's own Bias-T supply board connected to an external power supply) to the HackRF (see your post 3 days ago), then this will draw much more current than with the 50 Ohm resistor, because the base board mostly expects 12V and a higher current. That will surely destroy the HackRF Bias-T Output.

theRFnoob, 2020-06-17:
Thanks @Dirk275! Very good point that I might have easily burnt my bias-t circuits!

Dirk275, 2020-06-17:
Yes, it's with Mayhem.

gerryk, 2020-07-02:
My Aliexpress sourced HackRF + Portapack H1 works fine with Mayhem 1.1.1. Using an RTL-SDR LNA, I get about 30dB boost when I switch on the Bias-T vs. switched off.

Cidrch, 2020-08-04:
Hi. My portapack. Power is activated during Tx transmission (should be during Rx reception). Any firmware, both out of the box and mayhem. +DC 3.16v

berkinkartal, 2021-01-10:
Dear All,
I'm observing the same behaviour as RFNoob in regards to CLKIN port. CLKIN port leaks HackRF internal reference clk during normal operation and doesn't allow an external reference clock to be used (according to specs that should be a CMOS square wave clock, DC coupled). I tried this also with various firmware versions but no luck. Has anyone ever tested this?
Also I tried setting up a simple Signal generator using gnuradio. When using HackRF mode with portapack (v1.3.0), there's a little sinus clock sign on portapack screen, it always shows EXT (guess it refers to an external clock), although HackRF generates configured signal properly, eventhough without an external clock source connected.
Another observation is that if you connect this CLKIN port to a frequency counter or a scope with 50ohm termination or just connect a 50ohm terminator to it, clock source sign on portapack LCD turns to HRF (I suspect that means, hackrf internal clock).
And I couldn't manage to feed an external clock into hackrf so far.

Any ideas?

ArjanOnwezen, 2021-06-05:
Bias voltage not working looks like a hardware issue, since it is working for @gerryk  / @Dirk275  and others

closed 2021-06-05 by ArjanOnwezen

Stefan2483, 2022-06-09:
Bias voltage does not work for me with portapack H2 and Mayhem v1.5.4  I remember that it was working with an older version but I don't know the version number.... Could it be a firmware issue ? with H2 ?  or my hrf has a hardware issue as well....  If anyone else has problems with latest fw and H2 please comment. Thanks a lot for this firmware anyway... it is awesome

jLynx, 2022-06-09:
Are you able to test 1.5.3 and see if it worked there @Stefan2483?

Brumi-2021, 2022-06-09:
Hi @Stefan2483 , 
I checked  **fw  1.5.4.** , in both H1R2 ,  H2+ big CPLD, **and in both , bias voltage works flawlessly .**
Pls. make sure that 
* you activate it (example on top bar menu toggle option) (or I think also menu settings.) 
* and then test it inside  any receiver  app + Capture app.  (only it works in RX condition, that is correct) .

(it is true that there are some minor receivers, that we did not activate it , like  TPMS , ...because we are not aware about any LNA in that application. ) 

Pls. confirm and we will close that issue . , but I think firmware is correct. 
 Maybe you just checked  it on main menu , not inside any RX App.

Cheers.

Stefan2483, 2022-06-10:
Hello,

You were right I just checked again and bias voltage works fine on Receiving->Audio.
I think I was not checking it in the Receiving menu.

Thanks a lot for your feedback and sorry for my mistake.

Very good job guys. From my part there is no issue so you can close it.

Thanks,
Stefan