portapack-mayhem/mayhem-firmware #82: Speaker on statusbar not working ok / not in synch with headphones

euquiq, 2020-06-19:
The speaker icon in the status-bar on top of the screen is optional, mainly for H1 Portapack users who installed an internal speaker. You can toggle its presence from OPTIONS -> INTERFACE

When pressed, you mute (icon greyed out) or activate the speaker amp (icon green). 

Whilst this is happening ,the headphones output has a bug: Please get into receivers-> audio and tune an FM radio

1) **At first, status-bar speaker is greyed out** and everything works as expected:  Audio will go thru the headphones. **OK**

2) **Then you turn on the speaker (Green)**, it sounds BOTH on headphones AND speaker (two amps powered up!) **NOT SO GOOD** ... Actually, I think I read that particularly on H2 Portapack the screen goes dim, as there is some power issue!

3) **Then you turn off the speaker (grey it out)** and sound gets shut off from BOTH the speaker AND headphones. (BAD!)

The speaker mute / unmute is not working in harmony with the rest of the system. 

For H2 users this may not be a concern, since they wouldn't even place it on the STATUS-BAR for a start. But for the rest of us, this needs to be solved!

... Which I think I did: I will place a PR, shortly.

ArjanOnwezen, 2021-05-12:
@euquiq did you solve this issue with https://github.com/eried/portapack-mayhem/pull/83 ? If so @eried could you close this issue?

euquiq, 2021-05-13:
I solved it for H1, which can discern between speaker and headphones. On the H2 design this wont work, since audio for the speaker is crudely taken from the headphones connector and there is no software control over that.

So in short: If you want to keep the firmware "Pure" to official designs, which is the h1 portapack, then yes, it is "solved". But it won´t work on h2. So I can close this issue, as it is "unsolvable for the h2 users".

closed 2021-05-13 by euquiq