portapack-mayhem/mayhem-firmware #84: AM/SSB transmitters

AsciiWolf, 2020-06-26:
Please, consider implementing AM (DSB-AM) and SSB (USB, LSB) transmitters with microphone support into PortaPack Mayhem.

eried, 2020-06-26:
🤔 why those in particular? 

I do not think any of the currently active contributors is radio-savy to the level of implementing transmitters in an easy way for now. We are just learning about it. **But!** this is *your* opportunity to shine :P and [implement](https://github.com/eried/portapack-mayhem/wiki/How-to-collaborate#coding-new-stuff-or-fixing-bugs) them.

AsciiWolf, 2020-06-26:
Why those? Because they are commonly used by ham radio operators. :-)

I would try to implement them, but I am afraid that I don't have enough know-how and time to do it.

gerryk, 2020-07-02:
One place that one could start is here: https://github.com/df8oe/UHSDR/wiki/How-does-your-UHSDR-software-DSP-work
UHSDR is the firmware for the mcHF Software Defined Radio. The code is fully open and the documentation is pretty extensice. Not saying you could just lift code from one into the other but the implementation would have some similarities.

gerryk, 2020-07-02:
Actual audio -> IQ processing here: https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/drivers
/audio/tx_processor.c#L466

Which uses the conveniently provided ARM DSP floating-point FIR function: _arm_fir_f32()_ - https://arm-software.github.io/CMSIS_5/DSP/html/group__FIR.html#ga0cf008f650a75f5e2cf82d10691b64d9

I haven't been able to find equivalent DSP functionality in Mayhem code. I guess this is the main missing link.

Audio TX sample -> IQ processing is here: https://github.com/eried/portapack-mayhem/blob/master/firmware/baseband/proc_audiotx.cpp

johnelder, 2020-08-04:
I'll look at this when my time frees up a little later this week, unless someone else does.

texasyojimbo, 2020-09-16:
Looking at the code, I think the actual "guts" of the FM modulator are in /firmware/baseband/proc_audiotx.cpp

The logic of the execute method is basically this, I think:

Get the frequency of the audio sample, put in sample variable
Calculate "delta", by multiplying sample by fm_delta (basically this is the deviation)
It then does a bunch of stuff with sines and phases which I don't completely understand (although I think that it might technically be doing phase modulation rather than "real" FM here given that phase seems to be an important variable).

At any rate, it seems to me that it ought to be easier to write an AM/SSB modulator than an FM/PM modulator, since essentially all you need to know is the frequency and the amplitude. 

SSB Is really just sort of a translation of the baseband audio up in frequency (so a 1 kHz audio tone transmitted using SSB on 10 MHz becomes a signal with a frequency of 10.001 MHz). And AM is basically just mixing the audio with a carrier (so you have upper, lower sidebands and carrier -- 1 kHz audio tone results in 9.999, 10.001 and 10.000 signals).

gerryk, 2020-09-16:
You're basically correct in your analysis. You can read a decent article on the theory of implementing FM using an audio sample stream being transformed to an IQ sample stream here: http://ka7oei.blogspot.com/2015/12/adding-fm-transmit-to-mchf-transceiver.html

The very high level view of any IQ based transformation is similar...
1. get baseband sample
2. do 'some transform'
3. extract I and Q
4. send I & Q to transmit stage

the 'some transform' is the interesting bit. In other SDR s/w, to generate SSB, a FIR Hilbert Transform DSP function is used with +/-45 degree phase to give I (0degrees) and Q(+90degrees). I have not found this DSP function in the Mayhem (or upstream) codebases though, but it is available as part of the ARM SDK.

texasyojimbo, 2020-09-16:
Thanks Gerry. I think that the I/Q math (not necessarily coding ability) is
what separates the men from the boys on projects like this one.

TheMrRandomDude, 2020-12-09:
+support for this from me as well.

iz2fly, 2021-01-23:
2 million ham radio operators  worldwide thank you for your commitment so far, perhaps not all, and even more for adding USB/LSB transmission mode.

teixeluis, 2021-05-31:
I guess this issue can be closed, even though the current implementation seems to have issues, at least with some types of hardware.

Issue raised: https://github.com/eried/portapack-mayhem/issues/368

closed 2021-06-02 by eried