portapack-mayhem/mayhem-firmware #87: nodejs modul with protocol support for different 433mhz switches and weather stations for the RFControl Arduino library.

ea3iav, 2020-06-29:
I think this module was included in some havoc versions

**Describe the solution you'd like**
Maybe there is some protocol already working. Lole this one on Oregon products http://wmrx00.sourceforge.net/Arduino/OregonScientific-RF-Protocols.pdf

eried, 2020-06-30:
Can you point to the version that included this? 

Any colaborator here could easily merge that, but making that from scratch is not going to happen soon, unless you start fiddling with the code yourself 🎉 (but I know: [no time/no skills/blabla](https://github.com/eried/portapack-mayhem/wiki/How-to-collaborate#coding-new-stuff-or-fixing-bugs) 🤣)

eried, 2020-08-13:
I am closing this issue since there is another one with the same topic, with bounties on it already.

closed 2020-08-13 by eried