portapack-mayhem/mayhem-firmware #88: AIS time display error

vfatman0825, 2020-06-29:
[image]
The following minutes and seconds are displayed as a date

eried, 2020-06-30:
Interesting bug on the parser 🕵️‍♂️ thanks for the report :)

euquiq, 2020-06-30:
https://www.youtube.com/watch?v=WZQ_BjmpyqQ Here can be seen too

eried, 2020-06-30:
@euquiq found and fixed the problem immediately, can you check it @vfatman0825 ?
https://github.com/eried/portapack-mayhem/releases/download/nightly/portapack_nightly_202006302211.bin

vfatman0825, 2020-07-06:
> @euquiq found and fixed the problem immediately, can you check it @vfatman0825 ?
> https://github.com/eried/portapack-mayhem/releases/download/nightly/portapack_nightly_202006302211.bin

Thank you very much. This error has been fixed
[image]

eried, 2020-07-06:
Awesome!! @euquiq patch was already merged to master for future releases

closed 2020-07-06 by eried