portapack-mayhem/mayhem-firmware #91: wiki entry: Intended-Use-and-Legality

eried, 2020-07-02:
Thanks for the new entries in the wiki @gerryk! very helpful

I think it should sound more like an "user manual" than a blog post in particular in the header of this:
https://github.com/eried/portapack-mayhem/wiki/Intended-Use-and-Legality

What do you think? or what other people suggest?

i.e.:
"This is not intended to be legal info--"

I know other sections sound like a blog post still but they also need to be rewrited (the intro I wrote, or the "how to install",etc)

ps: I like also how you are using the graphics, consider that you can upload any image to the wiki, just using normal GitHub desktop

gerryk, 2020-07-02:
Hi Erwin...
Yes, I agree. To be honest, I was just typing in a sort of unconsidered
manner.
I will go back over it and make sure things sound a little more
'professional'.
I will also continue adding sections as time permits, and gratefully
receive comments, edit etc.
/ Gerry

gerryk, 2020-07-02:
Edits done. Please review at your convenience.

On the point of graphics, I note that some images appear as a white square rather than the actual icon in question.

/ G

eried, 2020-07-02:
Thanks Gerry :) the manual starts too look complete

closed 2020-07-02 by eried