portapack-mayhem/mayhem-firmware #92: Is there a way to make a custom splash?

Nuclrz, 2020-07-03:
Hi I just wanted to know if there was a way to make a custom splash image and what tools I would need.

eried, 2020-07-03:
Yes, you have to change `\firmware\application\bitmaps\bmp_splash.hpp` converting a compressed RLE BMP with https://github.com/eried/portapack-mayhem/wiki/Splash-and-other-images

eried, 2020-07-03:
BTW, I think it was 8 or 4 bit Compressed (at least in photoshop) but I am not 100% sure, try and please add that info to the wiki page in a new title if you succeed swaping the splash.

gregoryfenton, 2020-07-10:
I have written a simple guide at https://github.com/eried/portapack-mayhem/wiki/Create-a-custom-splash-screen

eried, 2020-08-10:
Nothing to add here. Closing the issue.

closed 2020-08-10 by eried