portapack-mayhem/mayhem-firmware #93: Creating a custom splash screen has an issue

gregoryfenton, 2020-07-04:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
Create a custom splash screen bmp_splash.bmp and convert it to .hpp using xxd

**Expected behavior**
The file should be drop in compatible with the code

**Actual behaviour**
The filename and the output file have the same name but the source code expects a different name. In this case the source is wrong and needs amending to prevent confusion.

**Affected versions**
All

**Additional**
The fix would be changing firmware/application/ui_navigation.cpp:
	portapack::display.drawBMP({(240 - 230) / 2, (320 - 50) / 2 - 10}, splash_bmp, false);
to
	portapack::display.drawBMP({0, 16}, bmp_splash, false);

I have changed the position code as my guide specifies a resolution (240x304).  It may be worth amending at a later point to check the resolution and automatically centre the image [x = 120 - (bmp_width / 2), y = (160 + 8) - (bmp_height / 2)]

gregoryfenton, 2020-07-04:
Just been playing with self-centring code, it's a trivial addition.

firmware/application/ui_navigation.cpp:
```
void BMPView::paint(Painter&) {
	/*
		bmp header format has 16 bit width at byte 18 (0x12), little endian
		bmp header format has 16 bit height at byte 22 (0x16), little endian
		In the code below:
		120 is LCD width / 2,
 		152 is (LCD height - top line height) / 2.
	*/
	uint16_t x, y;
	x = (uint16_t) (splash_bmp[19] << 8 | splash_bmp[18]);
	y = (uint16_t) (splash_bmp[23] << 8 | splash_bmp[22]);
	portapack::display.drawBMP({120 - (x / 2), 152 - (y / 2)}, splash_bmp, false);
}
```

eried, 2020-07-04:
Well... The tool is not intended to create drop-ins replacements or anything. If you see the code, also the size and position needs to be specified and the transparency so in any case the programmer needs to interact with it.

psyl0c, 2020-07-08:
is it possible to make it *.jpg *.png compatible ?

eried, 2020-07-08:
> is it possible to make it *.jpg *.png compatible ?

yes

gregoryfenton, 2020-07-08:
While JPEGs may be small they need to be expanded on loading which may take up too much memory at runtime.
For PNG, shouldn't be an issue.
However, why not simply use the ImageMagick convert function?
convert filename.png -type Palette -compress RLE -colors 16 BMP3:filename-4bit.bmp
before the xxd step?

While I type this it occurs to me that I should consider adding this to the guide, it saves a bunch of steps in the graphics program.

eried, 2020-08-10:
Nothing to add here, there is an entry in the Wiki now about the custom splash.

closed 2020-08-10 by eried