portapack-mayhem/mayhem-firmware #95: Request for mayhamn menu

psyl0c, 2020-07-06:
You should make survival mode / covid-19 mode
Maybe different modes with different menus:

Covid-19 mode ( where you can send and receive information )
Earthquake mode 
Tsunami mode

Hack Mode / Social Enginere Mode

Im just thinking outloud..

eried, 2020-07-06:
Send and receive to whom? If is a distress call and nobody is actually listening?

jwetzell, 2020-07-06:
What would a social engineering mode look like in the realm of an SDR?

psyl0c, 2020-07-08:
Erwin, I don't know, maybe a section for morse code:
like send morse code
decode morse code ( sound to text )
Make lights on 433 / 868 ( like in your video) turn on and turn off,  but with morse signals, you need to have all possible options, I want to do that to streetlights i Scandinavia , dude don't forget its MAYHAM out there, ppl is fighting over toiletpapper!!!

What if you enter the supermarket and you see 3032 million ppl standing in line, Just send out a distress call over the storelights to your buddy on the other side of the store to fetch you a 6 pack of TP, now you can survive the next week.

Im just saying this is a MUST have in the MAYHAM menu .

psyl0c, 2020-07-08:
Maybe we can use tha latest CVE-2020-6007 https://securityaffairs.co/wordpress/97392/hacking/philips-smart-light-bulbs-hack.html to philipe hue light bulbs over zigebee .

eried, 2020-07-08:
Yes but that is impossible to do in a reliable way. 

You can record for example the lights signals for your specific application and then use Audacity to create a interesting pattern for your SOS as explained here: https://github.com/eried/portapack-mayhem/wiki/C16-format so you can tell your buddy that there is people fighting over toilet paper.

What you are suggesting sounds a bit based on how hackers are depicted in movies, where no matter the environment there is a magical device that will find a way to do something. For now, the operator actually needs to do some of the work.

psyl0c, 2020-07-08:
hahah, nothing is imposible....

eried, 2020-08-10:
I am gonna close this issue now because it is not fully defined. Feel free to open a new request with a clear description about the request (and hopefully with the right name of the firmware in the title 🤣)

closed 2020-08-10 by eried