portapack-mayhem/mayhem-firmware #96: scanner improvement ( NFM + AM + ...)

marchelh, 2020-07-08:
Original question from Havoc ticket #363
alain00091 commented on 20 May

Is it possible to add a new demodulation to the scanner

1 maybe create 2 files "SCANNER_AM" and "SCANNER_NFM" and select which one we want to scan or

2 in the SCANNER file add at the end add the demodulation method like AM.NFM.SSB,, etc ... or
3 when we start the scanner ask which demodulation to use

marchelh, 2020-07-08:
I,m currently working on this item. This however is more work, because I want to implement this the correct way. Currently adapting the FreqMan code to support modulation from the frequency database files. Then adding these changes to the applications that use these files via FreqMan. My development tree already has the FreqMan working without the GUI part, which needs to be done because of compatibility. If this is done, I can extend the Scanner to have per channel modulation.

eried, 2020-07-08:
Awesome, can you please try to split the functionality tho? like first a PR for the freqman?

There is another PR pending for the freqman but it is being "polished" still :)

marchelh, 2020-07-08:
Eried, who is working on that FreqMan code? And indeed the FreqMan must be done first, because the scanner is using this for the frequency database. There is also still a bug in FreqMan, which I discovered yesterday, I will make a report for that one, and also try to fix this.

eried, 2020-07-08:
It is @alain00091 take a peek here: https://github.com/eried/portapack-mayhem/pull/80

Maybe you could do an epic fusion 😜
[image]

marchelh, 2020-07-08:
Yeh, that was stupid, I didn't check the pull requests. I will give it a try.

alain00091, 2020-07-08:
Hello, please update me about the bug, I 'm finishing freqman and scanners maybe tomorrow morning
Cheers

marchelh, 2020-07-08:
What on scanner and freqman are you finishing? I mean what functions or bug fixes?

alain00091, 2020-07-09:
Hello I just published it in my fork
And I need comments ( positive or negative)
Cheers

alain00091, 2020-07-09:
Hello Eried,

Can I upload to you or other my binary code,
It is based on your release MAYHEM 1,1.0, I rename it 2.0.0. Have not time to check it on 1.1.1 
I worked a lot and need comment .
The big issue was to fix a problem with Freqman, if there is special caracter NULL ( HEX = 00), it could crash ...
I hope I put everything on my fork, have a look
The only issue is on Fileman, I fix the crash if no SD or no file on SD but can not go directly to the modal view like Freqman, even if I clone exactly Fileman with Freqman, I fo first on the usual menu without any info, and then if you type the EXIT button we go to the modal screen

eried, 2020-07-09:
You can upload here dragging, or better in our discord :) we have a channel for letting people test-drive firmwares

alain00091, 2020-07-09:
how to get to the "discord
"

eried, 2020-07-09:
Links on the main readme (chat badge or in the text)

closed 2020-08-13 by eried