portapack-mayhem/mayhem-firmware #97: Problem upgrading the firmware

auskimus, 2020-07-08:
**What is happening?**
Describe here why you are unable to upgrade the firmware. Before describing your problems, please read: https://github.com/eried/portapack-havoc/wiki/Update-firmware.

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above if this is your first time.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
Add photos and videos of your procedure.

auskimus, 2020-07-08:
[image]
'hackrf_update.exe'is not recognized as an internal or external command,
operable program or batch file.
using  PortaPack H1 on Windows 10, tried 2 PC and multiple USB cables, unfortunately same result.

eried, 2020-07-08:
Unzip the files to a folder, do not run them directly from inside the compressed file

auskimus, 2020-07-08:
thanks for the prompt reply,  file is already unzip by 7z, i did follow the exact procedures on video and wiki

eried, 2020-07-08:
And you still get that error? can you record a small video of what you are doing?

auskimus, 2020-07-08:
Hello Eried while shooting a short clip of my misery suddenly Ta Ta its working, there was nothing different from the same procedures and steps taken before, thanks for your support and thanks for the great work.

closed 2020-07-08 by auskimus

eried, 2020-07-08:
😂😂 nothing different? cannot be, but nice that it works now

auskimus, 2020-07-08:
That happened while expecting it will not work, maybe the only difference between me and the others is that
I installed it fresh after erased the glitchy havoc firmware.

Kind Regards