portapack-mayhem/mayhem-firmware #98: Frequency Manager does not write in the shown database but instead in the first one from the list

marchelh, 2020-07-08:
The frequency manager is not writing the selected frequency file when you select a file(database) for adding an item. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open the frequency manager
2. Select a database
3. Add new item
4. Tab save
5. Item is added to ACARS file

**Expected behavior**
New item must be written to the selected database

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release
* Probably any

**Additional**
I only tested this once.

marchelh, 2020-07-08:
I will take this up for I'm working currently on the frequency manager.

alain00091, 2020-08-06:
Hello marchelh, I've just finished to fix a lot of problem with freq man

ArjanOnwezen, 2021-06-08:
Should be solved by  #372

closed 2021-06-08 by ArjanOnwezen