portapack-mayhem/mayhem-firmware #99: Wave signal generator app with Amp always on

ea3iav, 2020-07-11:
**Describe the bug**
When using the signal generator app, the output shows same result with or without amp turned on. As per other testing with anither unit on pc the ouput looks like the one with amp is on

**To Reproduce**
Steps to reproduce the behavior:
1. Go to signal generator app
2. On 2.4ghz which is around max output of the hackrf. Select for instance fm sine 
3. Look output with and without amp on. Both show same level around 16dbm

**Expected behavior**
There should be a difference on power level 

**Affected versions**
Mayhem 1.1.1

eried, 2020-07-11:
I guess this was not measured against the actual output right? (i.e. with another sdr to empirically confirm this)

ea3iav, 2020-07-11:
I used another hackrf unit on Pc using fm cw carrier on sdr Angel to stablish the base power output.
I came across this difference once I tested the portapack against these results i got on the other Hackrf I have
[image]

eried, 2020-07-11:
Great! could you please test this on other firmware versions? There was some change on portapack related with the amps but it would be difficult to find out without some extra tests.

If you can find any difference with this versions:
https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

That would help a lot.

euquiq, 2020-07-11:
I am testing it ala "old school", placing my portapack at some distance from my fm radio, tunning an empty frequency (i.e. 90.6mhz) and starting the Signal Generator (with the antenna closed in all its elements). 

With AMP 0 the wave can be listened but with lots of interference.
With AMP 1 the wave can be listened with no interference, loud and clear.

ea3iav, 2020-07-12:
Ok, how do I install the old legacy firmware? sorry

ArjanOnwezen, 2021-05-12:
@ea3iav Is this issue solved? If so this issue can be closed @eried

closed 2021-05-18 by eried