portapack-mayhem/mayhem-firmware #102: Problem upgrading the firmware

Tiodeer, 2020-07-14:
**What is happening?**
Describe here why you are unable to upgrade the firmware. Win 10 user. Regardless of the standard installer or DFU mode...neither can identify that the hackRF is actually connected to the PC...which is strange because not only does the device manager see it...the hack RF interfaces with SDR software perfectly. Ive even run the Zadig driver tool and rolled back the driver.

**Were you able to update the firmware before?**
I could upgrade from the vanilla porta pack to Havoc with no issue. 

**Can you try the upgrade with a different PC/Portapack/HackRF?**
I've tried five different cables...all of which could interface with Sharp/SDRConsole with no issues. I will try a completely different PC later.
[image]
 
Ive also tried 4 different USB ports. The SDR software can see the Hack RF...but the .bin somehow can not. Although the DFU method was able to connect...it only transferred 25k bytes and closed. 

**Additional**
Add photos and videos of your procedure.

euquiq, 2020-07-14:
This does not seem to be related to Mayhem firmware.

Tiodeer, 2020-07-14:
of course not. However, the installer can not see that the hackrf/portapack is connected is more likely the issue. I've tried this on an addition PC and the exact issue was duplicated.

jwetzell, 2020-07-14:
You mention the zadig driver tool, have you tried with the actual driver installer that comes in the firmware zip? Not sure if that will make a difference but something to try.

eried, 2020-07-15:
Hey @Tiodeer, can you check this video carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs

I think it covers 99.9% of the problems

har5ha, 2020-07-15:
I had the same issue last night, on my Win10 Pro[Version 10.0.18363.959] , I used the command `hackrf_spiflash -w portapack-h1_h2-mayhem.bin`  [hackrf_spiflash from PothosSDR/bin] instead of the Batch file which uses the hackrf_update.exe, internally.

eried, 2020-07-15:
> I had the same issue last night, on my Win10 Pro[Version 10.0.18363.959] , I finally used the command `hackrf_spiflash -w portapack-h1_h2-mayhem.bin` [hackrf_spiflash from PothosSDR/bin] instead of the Batch file which uses the hackrf_update.exe, internally.
> 
> I am on Mayhem 1.1.1 firmware now.

👀 there is another batch in the same folder to trigger the spi update, did you tried that?

har5ha, 2020-07-15:
> 👀 there is another batch in the same folder to trigger the spi update, did you tried that?

I initially tried "flash_portapack_mayhem.bat" but got the same error, "Error: HackRF not found" along with other batchfiles and they all have "hackrf_update.exe" in common. The "dfu_hackrf_one.bat" which was using the "dfu-util-static.exe" helped me to get through "HackRF" USB Not recognized issue, somewhere I messed up. Once USB was working I used `hackrf_spiflash -w portapack-h1_h2-mayhem.bin` and it upgraded!.

Tiodeer, 2020-07-15:
This community is SMART!!!

Ok...it appears that i have this working with the @har5ha method. I could not find my original spiflash .exe and simply found a replacement. I put (for the benefit of others) all of that into the installation folder. You may need to collect a few .dll files as well.
Once everything is in the same folder...run the command prompt:  hackrf_spiflash -w portapack-h1_h2-mayhem.bin

A warm thank you to everyone who provided input!  Now...who do i see about tweeking the audio receive parts of Mayhem?

closed 2020-08-10 by eried

MalyMali, 2021-11-22:
Ahoj. mě se stavalo to samé. Vyměnil jsem vodič (kabel k zařízení) a už mi to fungovalo skvěle.