portapack-mayhem/mayhem-firmware #103: How do I turn off the screen in HackRF Mode?

har5ha, 2020-07-15:
Hello!,

I am trying to turn-off the screen/display while in HackRF Mode(ie. when connected to a PC). I have the older HackRFOne (H1) model. The screen turn-off/timeout function works flawlessly inside Mayhem UI, though. I have had issues with Portapack screens earlier and would like to turn the screen off, when using SDR software on my PC.

Any help would be appreciated. 

Thanks in advance!

eried, 2020-07-15:
I have not tested this, but there is a parameter in hackrf_debug for that:
[image]

har5ha, 2020-07-15:
> 
> 
> I have not tested this, but there is a parameter in hackrf_debug for that:
> [image]

Thanks @eried  your suggestion worked. I used the > `hackrf_debug -u 0` [ie. Which is in my PothosSDR/bin folder]

Once again. Thanks for your help!

eried, 2020-07-15:
Cool! It cannot be triggered from the UI for now but it sounds like a good suggestion

har5ha, 2020-07-15:
> 
> 
> Cool! It cannot be triggered from the UI for now but it sounds like a good suggestion

Got you!. But I vaguely remember Havoc firmware having that functionality (I could be completely wrong). It's been a while I used my Portapack. Thanks!

eried, 2020-07-15:
If you can find an old version with this, we could easily integrate it :)

https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

har5ha, 2020-07-15:
Sure!, I will try searching for the version of Portapack[Havoc] which I suppose had it?, on my eternal HDD . Will get back to you.

sliceratwork, 2020-07-19:
I tried running `hackrf_debug -u 0` from the PothosSDR/bin folder, but the screen is still on.

Any ideas what could be the issue?

- Windows 10
- Genuine HackRF and Portapack running the latest Mayhem.

har5ha, 2020-07-19:
> 
> 
> I tried running `hackrf_debug -u 0` from the PothosSDR/bin folder, but the screen is still on.
> 
> Any ideas what could be the issue?
> 
>     * Windows 10
> 
>     * Genuine HackRF and Portapack running the latest Mayhem.

Did you enter the "HackRF mode" through Mayhem UI and get to the "Blue Screen", before you ran the above command?

Also check if hackrf_info command recognizes your device..

sliceratwork, 2020-07-19:
Yup, I tried running the command in HackRF mode.

Here's the result from hackrf_info:

```
hackrf_info
hackrf_info version: git-92ee144
libhackrf version: git-92ee144 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2927875f
Board ID Number: 2 (HackRF One)
Firmware Version: local-265a3537 (API:1.03)
Part ID Number: 0xa000cb3c 0x00554f5f
```

har5ha, 2020-07-19:
Hmm.. That's strange... The only thing I different was copy the "default.xsvf" from here [ https://github.com/mossmann/hackrf/tree/43e6f99fe8543094d18ff3a6550ed2066c398862/firmware/cpld/sgpio_if ] in "raw" format. And ran `hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf` As suggested here --> https://github.com/mossmann/hackrf/wiki/Updating-Firmware . That's about it. I can't guarantee it will work.. but you can give it a try..

eried, 2020-07-19:
Have you tried another version of the hackrf tools? I haven't tried the screen off function personally and cannot for a couple of weeks

sliceratwork, 2020-07-20:
@har5ha tried your suggestion, no difference.
@eried will try a different version of hackrf tools.

Will post an update here if/when I find a solution.

If anyone else could try disabling the hackrf UI and post their results here, that would be great. Might help other people having similar issues.

Thanks.

eried, 2020-08-13:
Any luck with a different version of the tools?

sliceratwork, 2020-08-13:
@eried nope. The screen is still on.

Will do some more testing tomorrow.

mcules, 2020-08-20:
In my hackrf_debug (v0.5) is no -u switch available.

eried, 2020-08-20:
The hackrf tools included on https://github.com/pothosware/PothosSDR are the ones I use. However we should do something about the tools versioning too :/

ArjanOnwezen, 2021-05-18:
Downloaded the latest version (2021.05.01)from: https://downloads.myriadrf.org/builds/PothosSDR/
Using Mayhem 1.4.0 
`hackrf_debug.exe  -u 0 `
will turn screen off
`hackrf_debug.exe  -u 1`
will screen back on again

So @eried  this issue can be closed

closed 2021-05-19 by eried