portapack-mayhem/mayhem-firmware #106: Where is the HackRF One Frequency Calibration Menu?

Dirk275, 2020-07-22:
**Describe the bug**
I cannot find the Frequency Calibration Menu for the HackRF One.

**To Reproduce**
Steps to reproduce the behavior:
Searched the UI menus, espacially "Settings".

**Expected behavior**
Under Settings (or as a separate menu point) you should find the Calibration Menu and be able to enter PPM values for calibrating the HackRF, which were stored in the internal memory for permanent use. This menu was part of the Havoc FW.

**Affected versions**
* Mayhem latest Stable Release 1.1.0
* Also tested: 1.0.3
* Maybe all versions

eried, 2020-07-22:
Do you remember something about the screen? I do not have my portapack handy right now for few more days, but if it was in settings AND it actually did something then it is still there. (only the fake menus and non working stuff was removed). Option 2) is in the options on the top (on the header)

euquiq, 2020-07-22:
I search "FREQUENCY CALIBRATION" on HAVOC's repository and cannot find a single reference to that menu. I frankly never heard of it. Can you please post an image showing the menu / app so we can pinpoint it inside furrtek's archived repository ?

Dirk275, 2020-07-22:
@eried and @euquiq ,
I'm looking for a picture of this menu on the net, but was not successful until now, but I'll keep on searching.
I remember it was very simple: a white field for the ppm number and the text "PPM" with the possibility to choose minus or plus to change the value. The sense of this calibration was to correct the internal TXCO base reference by receiving a station with good frequency constance e.g. a VOR and tuning this station to max. volume. The difference of the frequency of the max. volume to the nominal frequency can be used to calibrate the HackRF. With the former Havoc FW I had a value of 4 ppm as calibration value.
2) the function is/was not on the headline.

Dirk275, 2020-07-22:
Here:
https://github.com/sharebrained/portapack-hackrf/wiki/General-Use#frequency-correction
... is the description of the function and a screenshot.

euquiq, 2020-07-22:
I think you are referring to OPTIONS -> RADIO OPTIONS. If REFERENCE SOURCE is NOT external, but Xtal; then such configuration parameter is shown.

You can see it in: https://github.com/eried/portapack-mayhem/blob/master/firmware/application/apps/ui_settings.cpp

Specifically, from line 139 onwards:

```
	if( reference.source == ClockManager::ReferenceSource::Xtal ) {
		add_children({
			&labels_correction,
			&field_ppm,
		});
	}
```

You probably have an external clock (and there is nothing to calibrate).

euquiq, 2020-07-22:
> Here:
> https://github.com/sharebrained/portapack-hackrf/wiki/General-Use#frequency-correction
> ... is the description of the function and a screenshot.

Yep ... it is exactly what I am telling you on my message above. If  you are NOT using the crystal on your hackrf, but an external clock, you are not presented with the correcting field.

"The crystal frequency reference on the HackRF" is not used, if you have an external clock reference (in your portapack board).

Dirk275, 2020-07-22:
@euquiq,
I remember the Frequency Correction Menu was always active on Havoc, though HackRF uses the Portapack Xtal. I have used the function.
And that makes sense: the Frequency Correction should be possible for the Portapack frequency reference too. Of course the deviation is smaller.

euquiq, 2020-07-22:
@Dirk275 please get into the archived Havoc's code and check: The code I am referencing will be found in there too. You are referring to a very OLD OLD version, which did not take into account the external referenced commonly found today on portapacks (there is no deviation there). Cheers!

Dirk275, 2020-07-22:
Thank you, I will check!
Calibration always makes sense: actually my HackRF with Portapack is +22 Hz off at 120 MHz. Thats not much, but could be calibrated.

I don't understand: Why was this function on older and newer Portapack FW?
If a Portapack is installed on HackRF, then the Portapack frequency reference is used. If you are right, the Frequency Correction function would then always be deactivated.
But I have seen and used it ...

eried, 2020-07-22:
> I don't understand: Why was this function on older and newer Portapack FW?

Could you check older versions and how they work @Dirk275 ?

https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

If you load a very old one, check with a known signal and record the results we could investigate further and find what you are referring too, but as @euquiq says maybe it was always like it is now?

euquiq, 2020-07-23:
It was not "always" like it is now ... at first, on the first integrations / interactions between HackRF and Portapack, circa first months of 2016, the menu that @Dirk275 comments, surely was there.

But Portapack (even the original version, by now) has an "Improved tuning accuracy using a temperature-compensated clock reference (TCXO). The TCXO is accurate to better than 1ppm at typical operating temperatures."

(Text from [Sharebrained store](https://store.sharebrained.com/products/portapack-for-hackrf-one-kit))

So there is absolutely no need to "calibrate" anything. 

As further proof: That calibrating field (available if no external clock is present) uses an uint8_t (8 bits integer from -126 to 127 (or whatever). Ther eis no way to calibrate "sub integer PPM" which now is the actual accuracy of external clock itself).

Meaning: The external clock is -as seen on sharebrained store- way more accurate than whatever that field calibrates.

cheers!

Dirk275, 2020-07-23:
Yes, you are true. The PPM calibration used before with an 8-bit Integer will not help using it for calibration of a system with a 1 PPM crystal.
I'm trying to find out, where I know this menu from (I never used older FW, my first was Havoc of the end of 2019) and I definitely used that menu.
2nd I try to find out, why this PPM calibration existed in the Portapack FW: it would have been useless from the beginning, because also the 1st original Portapacks had 1 PPM crystal.

@euquiq , do you think it is possible, that Havoc FW of the end of 2019 didn't switch to the external crystal on Portapack automatically and therefore the Calibration Menu was active?

euquiq, 2020-07-23:
> @euquiq , do you think it is possible, that Havoc FW of the end of 2019 didn't switch to the external crystal on Portapack automatically and therefore the Calibration Menu was active?

Yes, there seems that it was a bug. BUT ALSO at first, Portapack did not include the high precision crystal (I Don't know if the original one was advertised as 1ppm but gave far worse results, or something like that). Which is why all the chinese selling website make such a fuss advertising the "NEW IMPROVED HIGH PRECISION CLOCK" And even sell those as an external "pluggable" module for existing hardware. 

I am sure that if you search inside furrtek and jboone github's about this, or even use google and invest some minutes, you may reach into the actual discussions and timeline of the whole thing, kind of a "hackrf portapack historian". I did it once on this topic, and I recall that the info "is out there". But being pragmatic, the code you see now (on mayhem, havoc, and even vanilla Portapack, is just a "well walked and discussed evolution" (not a bug). 

HAVING SAID THAT, if your portapack is not centered on it's frequency, and you can confirm this through some experimentation that may be reproduced by other users, it is perfectly possible to re-introduce the "frequency calibration" option. Thing is, I am afraid that uint8_t based calibration (integer, from -126 to 127) may not be enough ?? Sorry my knowledge on this topic is almost none.

eried, 2020-07-23:
Is quite strange that is 22 Hz... I played with those cheap usb tv dongles and I remember them having at most +/- 27... And those are COMPLETELY uncalibrated, so I would guess that even the chinese hackrfs should be next to spot on

Dirk275, 2020-07-23:
@euquiq 
if you would implement a frequency correction function for HackRF using a 1 PPM crystal, you would need a deviation of about ...
+- 100 Hz @ 100 MHz
+- 1000 Hz @ 1 GHz
+- 6000 Hz @ 6 GHz (max. frequency)
So a int16_t should do the job easily.

Dirk275, 2020-07-25:
Now I have looked into the code a bit deeper:
The Frequency Correction menu is located (labels, text) in ui_setting.hpp, the function in ...cpp.
The ppm variable is int8_t and in SetFrequencyCorrectionModel the value is devided by 1000, and later multiplied with 1000 (?). 
So the entered PPM value is perhaps used as frequency correction value (PPM × 1000), kind of PPk (parts per 1000).
I have not found the code, where the correction value is indeed used for correcting the receiver/transmitter frequency.

robots, 2020-10-13:
correction is applied in application/portapack.cpp:
`clock_manager.set_reference_ppb(persistent_memory::correction_ppb());`

And very important comment is in clock_manager:
void ClockManager::set_reference_ppb(const int32_t ppb) {
  /* NOTE: This adjustment only affects PLLA, which is derived from the 25MHz crystal.
   * It is assumed an external clock coming in to PLLB is sufficiently accurate as to not need adjustment.
   * TODO: Revisit the above policy. It may be good to allow adjustment of the external reference too.
   */

So the code is not ready to adjust PLLB. The change would be pretty easy, just copy the same code and calculation for PLLB and external reference frequency.

zigad, 2022-05-04:
Hi guys! My Portapack H2+R1 is also off by a few HZ. If I broadcast on 433.3300 MHz via HAM radio I can receive it on 433.3336 MHz that means it's off for like .0036MHz
[image]

I also tried for TX and RX on different sources. Would be awesome if someone could implement manual frequency correction.

xjols, 2023-01-25:
Hello!
I have a Portapack H2+ with teorical 0.5ppm TCXO clock. I'm now applying -10.45ppm offset calibration in Windows Console SDR software working as HackRF with USB. The same offset is present when the Portapack is working alone.
I think that the internal calibration will be usefull also when external TCXO is present in Portapack card. The promissed ppm is not correct in my clock!
Also very useful option will be an option to chose external or internal clock.
Thanks

gullradriel, 2023-05-20:
Frequency correction in Hz is available.
This solves the issue.

closed 2023-05-20 by gullradriel

gullradriel, 2023-05-20:
The ppm thing is used when in audio app i.e you want to listen something offset the center 
I added the missing set frequency correction.