portapack-mayhem/mayhem-firmware #107: Tx AIS Boats

vindn, 2020-07-26:
Using this current firmware, is It possible transmit AIS Boat messages?

euquiq, 2020-07-27:
Nope. No AIS-TX app present on current firmware.

vindn, 2020-08-09:
is it possible implement this feature? somebody thinks implement this feature?

eried, 2020-08-09:
Not that I know. But you can place a bounty to convince coders.

marcelrv, 2022-07-22:
I'm also interested, added to the bounty $10

Brumi-2021, 2022-08-27:
Hi , @vindn and all,  i also feel very interesting feature !  
But I can see a basic synchronisation time problem that needs to be solve before  . AIS is assuming that each station has a built-in  GPS receiver , that supplies  beside the position, COG, ... the exact GPS UTC seconds time.(with extremely high accuracy) .  And in any minute , AIS standard   define  2.250 slot messages .(0..till 2249) and the slot 0 always should  start  in the exact UTC minute . And the 2249 should end when changing the next minute. (And on top of that AIS is using 5 types of protocols to manage the slot allocation to every station , all of them variations of TDMA : RATDMA, SOTDMA, ITDMA, FATDMA, CSTDMA...)
To generate and transmit the different type of simulated  messages asynchronus, with  random TX  time should be feasible (similar as ADS-B messages or all other TX apps, obviously we should change the frequency , the modulation type, the data rate to 9600 bps ...) and those messages will be decoded by another Portapack with AIS RX App . But any official certified AIS equipment installed in a vessel  will not decode our asynchronus messages if we are not starting our transmission in the exact proper  window slot timing.  ... therefore I do not know how we can get before that exact clock UTC time in the Hackrf    , example synchronize before using AIS  our internal RTC with exact UTC time from DTC-77 in EU ?  or any other radio clock Transmitter service?  (because I am assuming that we can not send from laptop that clock through USB ...) ... this needs to be considered to make a real good AIS signal message simulation.... we just need to know very accurately each exact UTC minute starting time .

vindn, 2022-08-29:
Hi @Brumi-2021, did you see how AIS-TX (https://github.com/osmocom/ais-tx) works? In the AIVDM_Encoder.py you can specify several parameters but I don't know how it's solved this problem in AIS-TX. But like a first version without synchronisation, I think is good.

marcelrv, 2022-08-29:
@vindn AIS-TX does not deal with any synchronization. It just send at random timeslots.
I would also be fine with initial version in mayhem just capable of sending ais messages (like 18 & 24ab).

I think it should technically be possibleto implement with hackRF some sort of CSTDMA (Carrier Sense Time-Division Multiple Access) like many class-B devices implement. Just by listening to other AIS messages and using those to sync

Brumi-2021, 2022-08-29:
Hi @vindn , @marcelrv and all . 
I will also have a look into  AIS-TX example  and GNURADIO to get more familiar, and imagine the proper suitable  GUI . 

I feel it is really an interesting app to have it , and good chance to continue learning about Hackrf . 
But I am still learning , and I think , I  am  not the best one  to develpp it.  But on the other hand , we have many good TX App references . . Therefore sure possible to create it ...   Anyway, in any case  , I  will contribute to its development , debug and test  as much as possible as i can. 
 
Let's see if there are also  some other skilled developers willing to create it .

 But  do not worry , if nobody wants to start  , I would consider to start  it , but I will go slowly...  
Cheers

marcelrv, 2022-08-29:
I did some early steps into this as well. 
Not focusing at all on the GUI side of things, but instead on the signal handling of the matter. Esp the gfsk modulation I still don't know how to tackle.

I indeed see it as a good way to learn a bit more about the development side of mayhem.
I was able to send messages with gnu radio & hackrf but mayhem is a bit more difficult beast to tackle.

Brumi-2021, 2023-03-05:
Hi https://github.com/heurist1 , 
I could also see that sdrangel has an excellent AIS TX plug-in . 

I believe  that  we can inspire about that porting a lot from sdrangel , in terms of all steps , generating frame message,  NRZI encoding , Gaussian pulse shape filtering , FM modulation ,... 

I can try alone  about it , but I will go so slowly, and  I guess that If you (or any skilled developper  ) can  help me to create the fw files architecture structure and start about it on mayhem  , I will try to complete it much faster , 

I would be very happy to get some advanced help on it 
It is an open invitation, feel free to join .
Any help would be welcome ! 

cheers

eried, 2023-12-06:
@vindn probably is too late, but if you can ask for your 15 USD back, do it. Check -> https://github.com/bountysource/core/issues/1586

closed 2023-12-06 by eried

huseman21, 2025-06-30:
Yes lets get this working.  I think it could be the first actual "app" that could be used as a functional safety tool.  Add a small amp and everybody could see me at night.  Rivers and barges are dangerous.  Include a boundary alarm and its golden.
.