portapack-mayhem/mayhem-firmware #109: Screen sensitivity

smannchen, 2020-07-29:
Screen sensitivity should be increased, and and calibration tool seems inaccurate. Buttons especially in the corners don't feel as reactive as they should be.

euquiq, 2020-07-29:
I have the same problem on my (chinese clone) h1. It would help if we get more feedback on this issue from other H1 users (either genuine or clones). Thx!

eried, 2020-07-31:
Yeah, sadly (lol) my H2 touchscreen is fabulous and I dont have an H1 to test. But @euquiq could be your saviour

euquiq, 2020-08-02:
An idea would be: If you suffer this problem, please message me on Discord, so I can give you a test-drive firmware with different sensitivity values. We may investigate if this is a simple "switch" (like: Sensitivity: NORMAL / SPECIAL) or we rather need to place a variable input selector (like 0 - 100) so each one can fine-tune the best value for his / her screen.

smannchen, 2020-08-02:
Is it possible to put a sensitivity adjustment in the calibration menu?

eried, 2020-08-03:
Yes it is possible, but as @euquiq said it requires a lot of test first.

euquiq, 2020-08-14:
> Is it possible to put a sensitivity adjustment in the calibration menu?

I did just that!: When you enter into calibration screen, you can choose between STANDARD, ENHANCED and EXTREME sensitivity. 

Your choice is applied "on the fly" into the calibration routine itself too, so you dont have to break the screen trying to hit those calibration marks.

Once you calibrate, when you save the calibration data, the sensitivity mode will be saved  along with it, and be applied system-wide.

It is "ready" to be tested. Maybe I can issue a PR and a public test-drive version. It would be interesting to see how an H2 reacts with Enhanced and Extreme sensitivity. I would think they will not notice much.

gullradriel, 2023-05-10:
I think so too. Let's close it :-)

gullradriel, 2023-05-10:
As always when there is a bounty I don't know how it works.
Let's poke @eried :-)

closed 2023-05-10 by gullradriel