portapack-mayhem/mayhem-firmware #111: Bounty: Interactive bitmap titlebar

eried, 2020-08-01:
**Describe the solution you'd like**
The current header has two sections. The right side includes system icons and status (like ext antena and such), the left side has the back button (if needed) and the titlebar and version number:
[image]

This request is for replacing B (the titlebar) with a button that contains a graphical title:
[image]

When that title is pressed, the original About screen is triggered, as show here (in white pixels, with black/transparent background):
https://github.com/eried/portapack-mayhem/blob/master/firmware/application/ui_navigation.cpp#L478

The solution should consider the keeping A (back button) and C (version) intact, so it should look like this:
[image]

**Additional**
This is a test of the bountysource.com platform to see how bounties work

eried, 2020-08-01:
Bounty for this: https://www.bountysource.com/issues/92195032-bounty-interactive-bitmap-titlebar I am not sure why the bountysource plugin is not working -_-

klockee, 2020-08-02:
Issue should be solved by this PR: https://github.com/eried/portapack-mayhem/pull/112

closed 2020-08-02 by eried