portapack-mayhem/mayhem-firmware #113: rtl_433 implement

schmiernippel, 2020-08-02:
Implementing rtl_433 would be really nice.Dozens of ISM Sensors are in the neighborhood to be read out,so you have a wheaterstation in your pocket :-)

jamesshao8, 2020-08-18:
Which are you actually trying to communicate with rtl_433? I checked the code of rtl_433, the demodulation isn't complex, just FM and AM, which is already implemented in BLE and Analog TV receivers. The problem is there are too many devices to support.

eried, 2020-08-18:
I guess the backers of the bounty could provide samples and tell which devices they think this should support. I think if you click "solve" here https://www.bountysource.com/issues/92212604-rtl_433-implement they will get an email with "somebody started to work on this"

schmiernippel, 2020-08-18:
> What are you actually trying to communicate with rtl_433? I checked the code of rtl_433, the demodulation isn't complex, just FM and AM, which is already implemented in BLE and Analog TV receivers. The problem is there are too many devices to support.

But this is the main reason why this program is so nice because there are so many sensors supportet and frequently new added ! 
I use rtl_433 mainly for troubleshooting and exploring new weatherstations and temperature sensors(pool, indoor...)

jamesshao8, 2020-08-18:
I can try to decode some of the wireless gadgets, but don't have that much time to adapt everything. The weather station you mean is some kind of temp sensors at home? In my country, these kinds of device are using zigbee or wifi. I need to find something else to test.

helmarw, 2020-08-24:
as i postet [here](https://github.com/furrtek/portapack-havoc/issues/349) there is a lot of functionality missing in the TPMS monitor, even standard Schrader TPMS are not working fully... with rtl_433 no problem at all
so i think such an implementation would be beneficial ...

eried, 2020-08-24:
> I can try to decode some of the wireless gadgets, but don't have that much time to adapt everything. The weather station you mean is some kind of temp sensors at home? In my country, these kinds of device are using zigbee or wifi. I need to find something else to test.

What kind of recording will benefit you @jamesshao8? I mean, of course people will want "everything" but that will never be possible, so:

it would be nice maybe to have something similar the freq manager where you have a TXT to define what you decode? I am not sure how many different devices use different 433 protocols. 

**ALTERNATIVELY** make a base system that works for 2-3 things and write a guide how to write new entries with Radio Hacker and HackRF so everyone could contribute with their hardware?

jamesshao8, 2020-08-24:
I checked the supporting list of rtl_433, but there is no familiar device I can play with. 
And there is not much youtube video talking about using of rtl_433. I think it can start with some device that is common in China and Europe.

Yes, I do think making example for other's to add is a good idea. That's the benefit of making things open source.

eried, 2020-08-24:
Yes 👍 simple app + guide how to add more will get everyone integrating their 433 / (866?) devices.

jamesshao8, 2020-08-25:
@helmarw How do you trigger the sensor to send data? I think the sensors are sending data frequently only when the car reaches 80km/h. That's what I am experiencing on the road.

helmarw, 2020-08-25:
it needs to reach at least 30kmh not 80 or you send a 125 KHz tone to trigger the device or you have a the Miller Tool 9936
which can send this trigger signal and decode the TPMS bursts. But this is all written in the post i linked 5 posts above...
if you dont have neither you could build a rig with a motor spinning the sensor at this speed, doesnt need to be in the car 
nor does it need pressure to work...

jamesshao8, 2020-08-26:
[image]
This is what in my call. Are you familiar with that? @helmarw

jamesshao8, 2020-08-26:
It's strange that I can't found information of TTETP079 from FCC website.

helmarw, 2020-08-26:
i found it immediately ;)
[TTETP079](https://apps.fcc.gov/oetcf/eas/reports/ViewExhibitReport.cfm?mode=Exhibits&RequestTimeout=500&calledFromFrame=N&application_id=vY1BUf8a4T%2BM3T%2F64Rtppw%3D%3D&fcc_id=TTETP079)

you have to start search from this page:
https://www.fcc.gov/oet/ea/fccid

but i dont know exactly what type of sensor that is. you can search for the number [3641110AK80XA](https://www.google.com/search?client=safari&rls=en&q=3641110AK80XA&ie=UTF-8&oe=UTF-8) you get
some aliexpress but also schrader. if you have an rtl-sdr you can try rtl_433 to find out the exact type

thesle3p, 2020-12-07:
https://github.com/merbanan/rtl_433/blob/master/src/devices/simplisafe.c <-- would it be possible to implement this in Mayhem? I have an env to test it if needed.

schmiernippel, 2021-06-15:
Any news on rtl_433 implementation...??

evilpete, 2021-10-04:
It would most likely be easiest to add support the rtl_433's [bitbuffer](https://github.com/merbanan/rtl_433/blob/master/src/bitbuffer.c) call interface and decoder_output_data() hooks.
With these plus clear coding templates for importing [device](https://github.com/merbanan/rtl_433/tree/master/src/devices) code can made this into mostly simple busy work .

MKme, 2023-03-23:
Added another bit of $$$ to the bounty.  This would be an incredible add-on.

helmarw, 2023-04-04:
jupp agreed, I also contributed some $
would really improve the usefulness of this device

sorenrosenberg, 2023-08-24:
This would be an awesome addition. I want to contribute to the bounty as well, but is Bountysource safe to use? It seems abandoned and people are saying that the bounties aren't being paid.

eried, 2023-12-05:
Hi! anyone who added money to this bounty, please try to open a dispute on paypal, check -> https://github.com/bountysource/core/issues/1586

closed 2023-12-05 by eried

helmarw, 2023-12-06:
thnx, but that only works if its not older than 180days
everyone else will loose the money

eried, 2023-12-07:
yeah I know. I dont know if we can contact someone else, I added some money too back then. It is not a big sum, but it is a crappy situation