portapack-mayhem/mayhem-firmware #114: Feature suggestion: toy car hijacker

LordMcJefferson, 2020-08-03:
Hello,

I just want to suggest to have a look on mr Rgerganov's work:

https://github.com/rgerganov/rf-car 

Almost every toy car on Earth uses the same OOK codes to move. Seems quite easy to implement this in Mayhem and could be funny. Credits to mr. Rgerganov, of course.

eried, 2020-08-03:
Looks like a fun thing to add. But as always, the fastest way to get new features is to [code](https://github.com/eried/portapack-mayhem/wiki/How-to-collaborate#coding-new-stuff-or-fixing-bugs) it yourself.

Additionally, you could create a [bounty](https://www.bountysource.com/issues/92223922-feature-suggestion-toy-car-hijacker) for this and convince people to work on it.

jamesshao8, 2020-08-07:
there is an ook app in transmitter. why don't you use that to control your rf car?

LordMcJefferson, 2020-08-07:
> there is an ook app in transmitter. why don't you use that to control your rf car?

Because it can just send ONE code, so it would just power the car's engine for a split second. Even if you loop the code, the car would smash against the first wall towards it. That's very different from controlling the car imho :-)

I have some ideas on how it could be developed on Portapack, I already tried to code it myself, but the lack of documentation is frustrating...

LordMcJefferson, 2020-08-07:
According to mr Rgerganov, there are eight codes to move the car. Each direction is identified by a certain number of short pulses, with four long pulses in between to separate each code. For example, to move the car forward, you have to send four long pulses followed by ten short pulses (or 10 short pulses followed by 4 long pulses). Other directions are achieved by increasing the number of short pulses. We could map those codes to eight softkey on the touchscreen, set the frequency (usually around 27 MHz or 49 MHz, but there are some even in 2.4 GHz), push the softkeys to start loop transmission of the appropriate code and release it to stop transmission.

eried, 2020-08-07:
> I have some ideas on how it could be developed on Portapack, I already tried to code it myself, but the lack of documentation is frustrating...

Well this is one of the objectives of the project. The wiki is open to anyone for edits https://github.com/eried/portapack-mayhem/wiki so if you learned something about the OOK app or something while developing your attempt please write it there :) 

About the request, that is why I added a small bounty because it seems simple/fun/doable to create. You could try to convince more people to add to the bounty too and that will certainly increase the possibility of someone coding this :)

jamesshao8, 2020-08-12:
Hi @LordMcJefferson 
I have successfully used OOK to control my rc-car. But when I was analyzing the signal from ook app, it's very strange. If using gnuradio+hackrf to send the same waveform, it isn't strange, quite similar as original controller.

Waveform from orignal remote controller
[image]

The data in selection is: 1011110000100000101111000010000010
And I found  "1011110000100000" is enough.

So I edit waveform in ook app in portpack to be like this:
[image]

When I hit start button, the car react to my command indeed.
But when I use another hackrf and gnuradio to analyze what's sending out from portapack.
It's like this:
[image]

How's your result?

jamesshao8, 2020-08-12:
in /firmware/baseband/proc_ook.cpp

	if (cur_bit) {
			phase = (phase + 200);			// What ?
			sphase = phase + (64 << 18);

			re = (sine_table_i8[(sphase & 0x03FC0000) >> 18]);
			im = (sine_table_i8[(phase & 0x03FC0000) >> 18]);
		} else {
			re = 0;
			im = 0;
		}

I think this is what make waveform strange. The sine wave of OOK should be produced by LO as hardware, instead of software.

LordMcJefferson, 2020-08-12:
Guess you have to change the OOK type transmission. I would go for the 16bit type as it has more bits to set (but still not enough to move the car not forward only, always according to rgerganov's work). However, I wouldn't expect so much difference too.

jamesshao8, 2020-08-13:
I have already controlled my rc-car with that 526E type. My point is the waveform sents out from portapack is very strange, dosen't really look like ook to me. 16bit type is also strange.

eried, 2023-12-05:
Hi! @LordMcJefferson please try to open a dispute on paypal (if is still possible), check -> https://github.com/bountysource/core/issues/1586

closed 2023-12-05 by eried