portapack-mayhem/mayhem-firmware #115: file manager: Renaming and saving

smannchen, 2020-08-04:
**Describe the bug**
File manager saves in root of SD card instead of original location

**To Reproduce**
Open a folder
choose file and select rename
save file

**Expected behavior**
File should be saved in original folder.

**Affected versions**
08/03 Nightly and possibly previous versions

eried, 2020-08-04:
Thanks for the report. The sd operations are quite delicate indeed. I wonder if it would be better to hide those functions until they work reliable (like the rename, wipe sd, etc).

I'll test your bug and try to find the cause first, tho.

smannchen, 2020-08-04:
Agreed, it's much easier editing the files on pc anyhow.

euquiq, 2020-08-04:
ouch, sorry, I went on and fixed it. Now I see you self-assigned the task

closed 2020-08-04 by eried

alain00091, 2020-08-10:
I did the same, I was not aware that it was already detected and fixed !