portapack-mayhem/mayhem-firmware #121: NOAA weather alert/SAME decoder

smannchen, 2020-08-07:
Would like to see an app for decoding SAME data from NOAA weather radio. Messages should be presented in a simple easily readable way. Also the ability to switch to the audio app directly from that screen would be a plus or if possible to just have the audio already available and just unmuted by the app when an alert is received.

jamesshao8, 2020-08-12:
I have read some code and have some idea to decode noaa image, it's fm + am, should be easier than sstv. But no idea for the message.

eried, 2020-08-12:
You can try to write the decoder and post it in the #test-drive channel over in discord

smannchen, 2020-08-12:
Wasn't really thinking a Sat decoder, just a weather alert decoder like the ones the nws uses on weather radios, but decoding the text as well. However now that it's been mentioned this is an awesome idea as well. I think it's more than capable of decoding the sat images. Probably no harder than the sstv.

smannchen, 2020-08-12:
I will edit this to specify, and you can open an issue for noaa sats if you like.

texasyojimbo, 2020-08-14:
Looking at the technical specifications, SAME is transmitted using AFSK, at 520.83 bits per second, with a mark tone of 1562.5 Hz and space tone of 2083.3 Hz.  Individual bits last 1.92 ms. Each byte is eight bits with no error correction, and the most significant bit is always 0 and transmitted last. Each message is repeated three times in lieu of error correction, with the receiver intended to "vote" (best two out of three) to determine correct bit value. 

https://www.nws.noaa.gov/directives/sym/pd01017012curr.pdf

https://en.wikipedia.org/wiki/Specific_Area_Message_Encoding#History

It might be possible to modify the existing AFSK decoder to work for this, though my perception (not necessarily fact) is that the AFSK decoder is sort of broken right now.

euquiq, 2020-08-14:
I don't think AFSK is broken. I did look today into the SAME specs, etc, but sadly enough I have no NOAA / SAME down here to use for testing. Anyone in USA, did try to get into AFSK modem, configure its parameters and step into the right frequency ? I assume so. No SAME message is received ?

texasyojimbo, 2020-08-15:
Thank you; I may be mistaken re: AFSK and just not using it correctly (not
sure we have documentation).

Since it's AFSK, we ought to be able to use an audio sample for testing.

For example: https://www.sigidwiki.com/wiki/Emergency_Alert_System_(EAS)

gschwiter, 2020-11-25:
I would like to see the SAME codes as well. I have this on my scanners however it would be nice to have this while on the go. i would also like to see it save the messages as some disaster events can have several messages.
thanks

cql1983, 2022-04-22:
It seems that James shao8 has implemented this function。

https://github.com/jamesshao8/portapack-havoc-modified

Brumi-2021, 2022-04-22:
> It seems that James shao8 has implemented this function。
> 
> https://github.com/jamesshao8/portapack-havoc-modified

Thanks for the link ! I could see the decode NOAA sat image decoded , but I understood , that  he mentioned not released yet in that repo , the code. 
in his readme , he mentioned 
" Special Firmware supporting SSTV and NOAA receiving"
...
Features above are not disclosed yet."

[image]

darknebular, 2022-07-21:
This app called NOAA Receiver Will be fantastic.
I hope, someday, this app will be included in Mayhem firmware.

Someone has tried the Shao8 firmware in portapack H1?

eried, 2023-12-05:
Hi! anyone who added money to this bounty, please try to open a dispute on paypal, check -> https://github.com/bountysource/core/issues/1586

closed 2023-12-05 by eried