portapack-mayhem/mayhem-firmware #123: RDS TMC

nocomp, 2020-08-08:
hi folks,
never thought about implementing the TMC service as a TX or RX function?
TMC is the service that send traffic infomation to gps. 

more information here:

https://en.wikipedia.org/wiki/Traffic_message_channel
http://www.windytan.com/2013/05/a-determined-hacker-decrypts-rds-tmc.html

nocomp, 2020-08-08:
more docs
https://www.blackhat.com/presentations/bh-usa-07/Barisani_and_Bianco/Presentation/bh-usa-07-barisani_and_bianco.pdf

perl script that can be used as basis

http://dev.inversepath.com/rds/srdsd

eried, 2020-08-13:
Remember that you can "invite" people from the discord to work on this via [bounties](https://github.com/eried/portapack-mayhem#what-if-i-really-want-to-pay-for-something) 😜

Dirk275, 2020-08-16:
RDS and TMC data are transmitted mostly by FM radio stations. There is no connection to GPS.

nocomp, 2020-08-16:
Yes, i know thats why it shouldn t be too hard to port considering that rds is alreay functional for fm radio. Just need to adjust frequency ansd addbhex code for alert type, i think