portapack-mayhem/mayhem-firmware #125: File Manager rename file :  rename the file but put it always in ROOT directory

alain00091, 2020-08-09:
**Describe the bug**
File Manager rename file :  rename the file but put it always in ROOT directory, even if it was in a sub directory !

**To Reproduce**

go to File manager
select a sub directory
rename a file
the original file disappears from the sub directory
the renamed file appears in ROOT directory ( /)

**Expected behavior**

the renamed file should appear where was the original file ( same sub directory)

**Affected versions**
MAYHEM 1.1.1
maybe all

I'm fixing it

eried, 2020-08-09:
Nice @alain00091 ! Please create a clean PR with the fix :)

Remember:
* Create a new branch 
* The less amount of edits
* Only in the context of the fix
* Switch no another branch after finishing with the fix (to avoid appending more commits to the wrong branch)

Example: https://github.com/eried/portapack-mayhem/pull/86/files

alain00091, 2020-08-09:
yes  but in which branch     sharebrained or mayhem or my fork ?

eried, 2020-08-09:
Create one based on the repo you want to do the pull request, i.e. Mayhem in this case

euquiq, 2020-08-09:
This has been fixed 5 days ago !

https://github.com/eried/portapack-mayhem/issues/115

euquiq, 2020-08-09:
AlainD, you should get into our Mayhem Discord server. In there we are discussing every fix, doing test-drives, and talking about life and whatnot.  Here is an invite link (it will work for 24 hours) https://discord.gg/JPQWqA

alain00091, 2020-08-10:
Euquiq, the link is invalid and I do not know why

alain00091, 2020-08-10:
Now it works, I'm connected

alain00091, 2020-08-10:
Eried,

I couldn't make the PR on MAYHEM, dont know why
So I made it on my fork  

"File manager improvement when there is NO SD CARD or NO FILE on SD CARD"

eried, 2020-08-10:
Ok but that won't work. Anyhow you can chat on discord now, we can close this.

closed 2020-08-10 by eried

alain00091, 2020-08-10:
I am on discord now, could you please give me your Discord tag ?

eried, 2020-08-10:
Did you followed the invite? you will see all our channels and discussions there. Also in the main page of the repo there is a small badge that says "Chat" and amount of people connected that you can press to join too :)