portapack-mayhem/mayhem-firmware #126: Frequency generator issue

reclaudio, 2020-08-09:
Thanks a lot for this marvelous work !
I have recently bought an HackRF + Portapack 2 with v1.1.0 installed .
I have done some test on the Signal Generator section and looks very fine considering the costs involved .
I have noticed some problems : 
A) The output frequency doesn't go upper than 2700 MHz . You can write it , but can't exceed 2700 MHz , after it doesn't generate . Rotating the knob you can also  exceed 2700 MHz , but arriving at 2700 MHz it really stops there .
B) I wasn't able to understand how and if the BW setting change something , the modulation looks always the same 
C) I wasn't able to generate a pure carrier , if not using the Noise modulation that seems not to work , or using a Sine modulation with frequency = 0 
D) I didn't understand how to use the CW modulation . I tried to push the buttons ....

Is where something that I am doing wrong or some problems was corrected in the last firmware version ?

Of course the bigger limitation is point A . 

Thank you and kind regards .

Claudio Re I1RFQ

reclaudio, 2020-08-09:
I also checked  now that also that any other app , like analog receiver , analog transmitter and jammer doesn't work at frequencies higher than 2700 MHz .
Can someone help ? 
Thank you and kind regards .
Claudio Re I1RFQ

eried, 2020-08-09:
It sounds like a problem with your hardware. Do you have another way to test on that frequency?

reclaudio, 2020-08-09:
Thank you for your answer .

What you mean with the hardware ? 

The Hack RF ? The Portapack2 ? 

The instruments for the tests ?

I have tested all the other TX and RX applications with an HP8560E Spectrum Analyser and Marconi 3031 Signal Generator , that use every day and are both checked .

I also of course cross checked them together .

The limits are always 2700 MHz .

What about new Mayhem firmware or HackRF firmware ? 

A: eried/portapack-mayhem
Cc: reclaudio; Author

 

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

eried, 2020-08-09:
Can you transmit with the hackrf only?

reclaudio, 2020-08-09:
Don't know. Is there a Windows software with support for HackRF? I don't use Linux...

euquiq, 2020-08-09:
> Don't know. Is there a Windows software with support for HackRF? I don't use Linux... -----Original message----- From: Erwin Ried notifications@github.com Date: Sun, 09 Aug 2020 19:40:20 +0200
> […](#)

You can use HackRF with lots of windows Software. You have a faulty hackRF.

eried, 2020-08-10:
As @euquiq  said, almost all SDR Windows software supports HackRF. Try with SDR#

I'm gonna close the issue for now since it is not related to the firmware

closed 2020-08-10 by eried