portapack-mayhem/mayhem-firmware #128: Hack RF + Portapack 2 not working with Mayhem

reclaudio, 2020-08-10:
**Describe the bug**
Every app RX and TX doesn't allow to go higher than 2700 MHz 

**To Reproduce**
Steps to reproduce the behavior:
1. Go for instance to Utility  
2. Tap on Signal Generator - Sine - 2701 MHz - Start . It only generate spurious . It works correctly from 1 MHz till 2700 MHz , but not from 2700 MHz to 6000 MHz 
3. If I connect the hardware to  HDSDR in HACKRF mode it works fine , see two pictures attached with a Signal Generator at 3 GHz .

**Expected behavior 

The hardware is rated to  work till 6 GHz 

**Affected versions**
Firmware version v.1.1.0 received installed .

You would like to switch to Discord ? 

Thank you .

Claudio Re

reclaudio, 2020-08-10:
[image]
[image]

Sorry , forgot the images , here you have :)

reclaudio, 2020-08-10:
[image]
[image]
[image]
[image]

Here also you have some images of the signal gererator app , when tuned at 2700 MHz and 2701 MHz with the keyboard

reclaudio, 2020-08-10:
The picure of 2701 MHz was in start , because made before  I pushed the button : )

reclaudio, 2020-08-10:
In another thread I have seen around people complained not to be able to see WIFI at 5 GHz . coud be this the reason .... ?

euquiq, 2020-08-10:
Just to be on the same page, please CHECK the "AUTO" feature, so you know that when you change the parameters on-screen, those are reflected on the transmission status.

I will check this with my portapack and an SDR software with an rtl dongle. I also suspect that the guys with this potential problem, just got a burned IC on their hackRF

reclaudio, 2020-08-10:
Dear Erin ,

                 I have already checked the AUTO feature .

No way .

The same 2700 MHz issue is on ALL the TX and RX apps of the Mayem Portapack 2 . 

The HackRF that is in the package with Portapack 2 perfectly works driven with HDSDR till 6 GHz .

An SDR Dongle could arrive till around 2 GHz .

If you need to check over 2.7 GHz , you need a spectrum analyzer and or signal generator or at least an Adalm Pluto SDR or an HackRF driven with HDSDR to see the signals from HackRF + Portapack 2 .

If you need further checks from my side , please let me know .
I have another separate HackRF I bought when started on the market that again works perfectly with HDSDR . 

Thank you and kind regards .

Claudio 

A: eried/portapack-mayhem
Cc: reclaudio; Author

 

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

euquiq, 2020-08-10:
But those hackRF you mention that works fine .. which firmware do they have ?

reclaudio, 2020-08-11:
Joe can I check this ? :) . Thank you

reclaudio, 2020-08-11:
Sorry. How can I check this..

euquiq, 2020-08-11:
@reclaudio Eried just jammed a 5.6Ghz drone camera with his portapack and the TX-> JAMMER app. We've been discussing this on Discord. There is even a video of the ongoing jamming.

euquiq, 2020-08-11:
Now, we used the Tools -> Signal Gen, on that freq, to "jam" the 5.6 Ghz drone camera. 
 @reclaudio  ... You have a faulty unit. It is not a problem in firmware.

closed 2020-08-13 by eried