portapack-mayhem/mayhem-firmware #130: UI Improvement: Turn knob on speaker icon in status bar to adjust volume

xologram, 2020-08-12:
I think this would be useful feature. Click can stay mute.

eried, 2020-08-12:
There is already a speaker icon (hidden by default) for the H1 users to control the audio. So this would conflict with that 😱 but it is a good idea to have in mind. Maybe pressing up again on the header icons reveal a small volume knob...

xologram, 2020-08-12:
I was actually thinking of using that same speaker icon and when you have it selected I could turn the knob and set the volume, but when you click it - it behaves like it is now: mute. At least the icon on H2. Not sure if H1 icon is different. I *think* this way would be the easiest to implement. 

If this works, another one would be to turn the knob on sleep icon to modify LCD brightness.

on the other hand it would be great to have the ability to expand the header icons as I suspect we'll have more and more there with time. For example: one could be to jump to audio app on active frequency from any other app. just spitballing but you get the idea. 

btw, massive thanks for maintaining this!

eried, 2020-08-12:
Keep in mind that there are some side effects of using that with the H2. The screen dims when you turn it on and some people say it might burn the amp. 

That is why is only for H1. I would love to autodetect H1 from H2, but it seems not possible.

xologram, 2020-08-12:
FWIW i personally don't have any problems with that icon on H2. I don't have speaker hooked up, but I do use headphones and when I plug them in, icon stays muted but sound is on. Once I cycle the icon it starts behaving as expected with respect to mute. Zero issues with it - screen does not dim etc. I suspect that may be due to speaker drawing more power so it dims the screen and causes issues? 

My programming skills go as much as simple scripting but I wonder could you detect screen size and based on that assign model..

eried, 2020-08-13:
> screen does not dim etc. I suspect that may be due to speaker drawing more power so it dims the screen and causes issues?

Cool then. I am not really sure about the effects, but since in the H2 the switching between speaker and headphones is automatic, then you do not really need the speaker icon. When you toggle mute/unmute you do not perceive any change in the screen? even with the headphones plugged? 

> My programming skills go as much as simple scripting but I wonder could you detect screen size and based on that assign model..

Sadly, no. The software is not aware of the bigger screen (it is the same size in pixels) so it is just a "physical" magnification.

gullradriel, 2023-05-20:
We will not do it as it would mean going in each existing and incoming apps to check for a volume message and or monitor volume persistent memory settings to reflect in the GUI.
Use the volume control when it is available.

closed 2023-05-20 by gullradriel